ArduPilot/ardupilot #15308: Want to make devacopter

Moksh-2000, 2020-09-11:
Hello , i hope all are good ☺️

I want to make decacopter but I still can't find how to do that , (last 2 weeks)  , some people say only motormatrix.cpp custom on your choice and convert into .apj is it right ? 
Can I hope someone help me ?

Thank you 🙂

rmackay9, 2020-09-11:
@Moksh-2000, AP doesn't support decacopters at the moment.  It's not very difficult to add support but someone would likely need to do a little bit of development.

There's a video here from a pro user who made a custom motor matrix https://www.youtube.com/watch?v=c6aCwSHr_is although I suspect you've already seen this.

Moksh-2000, 2020-09-12:
> @Moksh-2000, AP doesn't support decacopters at the moment. It's not very difficult to add support but someone would likely need to do a little bit of development.
> 
> There's a video here from a pro user who made a custom motor matrix https://www.youtube.com/watch?v=c6aCwSHr_is although I suspect you've already seen this.
> 
> By the way, AP already supports dodeca-hexacopters which have 12 motors. https://ardupilot.org/copter/docs/connect-escs-and-motors.html#dodecahexacopter

Moksh-2000, 2020-09-12:
Sir this video already saw , and also comment on this guy on YouTube + our ardupilot discourse but no reply 😕 , sometime I'm very disappointing , and this topic great on 3weeks ago continuously I'm searching N searching When I got bored then reply to you.
Any help appreciated
Thank you 🙂

muramura, 2020-09-13:
I got curious about the decacopter.
I discovered the motor layout from the video.
[image]

I created a table based on this layout.
[image]

Moksh-2000, 2020-09-13:
Can you show my decacopter design ? , And yes I saw this video after I inspired 😉 , I'm also curious about how it's perform on 9450 props with custom design my motors 😎 , do you have any idea to put in pixhawk ? Because I already done but I nothing find on any suggestions. I am very happy that your message came.
Thank you 🙂

amilcarlucas, 2020-09-13:
I already explained exactly what lines to copy paste in the motors.cpp file and another user already told you how to change the code. The Wii explains how to compile the code from scratch.
We need you to tell us what exactly is not yet clear.

muramura, 2020-09-14:
@Moksh-2000 san.  I have finished implementing the sitl environment.  I have added a new frame class 14.  sitl is not aware of this class.  I contacted discord.

muramura, 2020-09-14:
I'm a source that has finished implementing the whole thing.
If you try to check the operation with SITL, the plane will become a plane.
I want to solve this problem.
Can you solve this problem?
https://github.com/ArduPilot/ardupilot/compare/master...muramura:AP_Support_decaCopter

Add DECAROTOR to the common.xml and minimal.xml of MAVLINK.
MAVLINK is another GITHUB.

[image]
[image]

amilcarlucas, 2020-09-14:
Very nice work @muramura . @Moksh-2000 can you test the code on your copter?

Moksh-2000, 2020-09-15:
@amilcarlucas I'm sorry for , im noob and also I don't have pc or laptop so I can't , all time pushing on Android as my pc , 
@muramura thank you soo much sir can you tell me how to download .apj file ? I will try on my copter , once again thank you ☺️

muramura, 2020-09-15:
@Moksh-2000 san. You have the actual device, right?
If you check the operation with SITL and it works properly, it will be the actual device.
I have not been able to check it with SITL.
Can you find the reason why the icon is plain in SITL?

Moksh-2000, 2020-09-15:
Sir as I mentioned i don't have pc or laptop so I can't check I'm soo sorry plz forgive me 😔 , i have ready my frame with preinstall pixhawk.
Thank you 🙂

rmackay9, 2020-09-15:
@muramura, maybe you could create a PR with these changes?  I can create a binary for @Moksh-2000.  It's an unusual frame but it's fairly straightforward and doesn't take up a lot of flash space to add support for it.

muramura, 2020-09-16:
@rmackay9 san. I will proceed with the PR work.

muramura, 2020-09-16:
I have added a 10-rotor helicopter type DERCAROTOR to the MAVLINK protocol.
https://github.com/mavlink/mavlink/pull/1479

muramura, 2020-12-11:
@Moksh-2000 san. 10 Rotacopters have been merged into the master. check it out and update us on the status.

rmackay9, 2020-12-11:
I think I'll close this issue now.  Definitely we'd like to hear back from someone who actually has the vehicle to test with.

closed 2020-12-11 by rmackay9