ArduPilot/ardupilot #15311: AC_WPNav: bad heading occurs during large vertical distance difference

lukezhqin, 2020-09-11:
#15310 Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[×] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
Same problem maybe occurs on other platforms also.

**Airframe type**
QUAD, hex, Y6, octa etc

**Hardware type**
Pixhawk, Cube

**Logs**
_log BIN file would uploaded later_

Please see some SITL screenshots here:
[image], 
[image], 
[image], 
[image]

rmackay9, 2020-09-11:
Thanks for this detailed analysis.  Top priority at the moment is getting s-curves into master and hopefully we will have a PR ready for review in a week or two.  I hope this will solve the problem albeit in a different way.

IamPete1, 2021-08-16:
I cant reproduce this on master, must have been fixed by Scurves. @lukezhqin please re-open if you can find a mission that will cause this.

closed 2021-08-16 by IamPete1