ArduPilot/ardupilot #15321: Roll,Pitch and Yaw axis are moving on Kakute F7 Mini V2 even when the board is stand still.

ninja-zx11, 2020-09-12:
Board is flashed with
 `commit 4ed486bb92f182f83034f35b65834ef596aba482
Author: Randy Mackay <rmackay9@yahoo.com>
Date:   Fri Sep 11 15:18:15 2020 +0900

    AC_AttitudeControl: revert Add PosControl PID logging

APMVERSION: ArduCopter V4.1.0-dev`

Pitch,roll and yaw axis readings are moving without moving the board.This is Kakute F7 Mini V2 board with MPU6000 IMU.I have not done the accelometer calibration yet.

Installed INAV and all axis are stand still and working fine.

[Attitude moving.txt](https://github.com/ArduPilot/ardupilot/files/5213178/Attitude.moving.txt)

[2020-09-12 12-01-23.zip](https://github.com/ArduPilot/ardupilot/files/5213180/2020-09-12.12-01-23.zip)

IamPete1, 2020-09-12:
1. Calibrate your sensors 
2. Go Outside and get a good GPS lock

The AP EKF fuses data from all sensors, so bad GPS = jumpy horizon.

The stable release if probably a better place to start.

closed 2020-09-12 by IamPete1

ninja-zx11, 2020-09-13:
Now Calibrated accelrometer and compass.Went outside and got good no. of sats with good HDOP and 3D fix but still having issue.
I am attaching the tlog and parameters.

Yaw and roll are moving quite a bit.
[2020-09-12 19-22-59.zip](https://github.com/ArduPilot/ardupilot/files/5213531/2020-09-12.19-22-59.zip)

vierfuffzig, 2020-09-13:
@ninja-zx11 without having looked at that log yet, do you see the same sort of horizon movement when you set INS_FAST_SAMPLE = 0 (and reboot) ?

ninja-zx11, 2020-09-13:
@IamPete1   Loaded Arducopter newest Stable V4.0.3 and issue is gone.So it must be something on the development version.

@vierfuffzig   I tried  INS_FAST_SAMPLE = 0 on the latest dev version but that didn't solve the issue.I am keeping  INS_FAST_SAMPLE = 0 on stable version.Do you think i should keep fast sampling disabled as this board has MPU6000?Thanks.

andyp1per, 2020-09-13:
@ninja-zx11 4.0.3 does not support fast sampling on the MPU6000, only 4.1 does. I am running with this board on 4.1 without issue, so I'm perplexed as to why you have a problem.

ninja-zx11, 2020-09-13:
@andyp1per   Yes I am curious to know the reason too.On the latest dev version,my roll was leaning to one side and even leveling via mission planner's level function could not make it level.

Can something be extracted from the tlog I attached?

vierfuffzig, 2020-09-13:
@ninja-zx11 that was just a shot from the hip due to a recent issue with at first glance compareable symtoms, caused by imu resets associated with fast sampling. it was on different board type and different IMU though. based on the additional information provided, yours most likely is unrelated.

andyp1per, 2020-09-13:
I have a bare board. Just calibrated and all looks fine - rock solid, no moving issues. I have a 3" quad with the same board - all fine. Are you doing this using USB power or battery? I have seen issues with peripherals taking too much power and that can cause problems for the flight controller. Are you able to try without devices attached? Are you using battery power? How hot is the board getting?

ninja-zx11, 2020-09-13:
I am testing with USB power and just GPS and compass connected.I can try going back to latest dev version again and see how it behaves.My board is getting warm but I have a heat installed on the F7 BGA chip.
Does you board have MPU6000 or ICM gyro?

andyp1per, 2020-09-13:
I have to identical 3" Geckos, one with a v1 (ICM) and one with a v2 (MPU). Both are fine. There are some issues with leans, but that's different to the problem you are describing here.

andyp1per, 2020-09-13:
Try calibrating on battery power and see how it behaves on battery power.

rmackay9, 2020-09-13:
I would try disconnecting the GPS just to confirm that this is the issue... I'm pretty sure that it's the GPS data getting fused into the EKF that is the cause as @IamPete1 says above.

ninja-zx11, 2020-09-14:
@rmackay9  I flashed latest dev version and disconnected the GPS and horizon was stable.Then i connected GPS and it started to move slowly.It seems that the unwanted movement depends on the GPS accuracy.Even outside,if you have little less reception,HUD would start leaning and Yaw starts moving around even during the presence of magnetometer.

I then flashed the stable version and Horizon and Yaw is  always stable,doesn't matter if the GPS is connected or not.And more important no leaning due to board temperature.

rmackay9, 2020-09-14:
@ninja-zx11, OK, thanks for the report.  We have been making various fixes and improvements to the EKF since the Copter-4.0.x release so I suspect it is related to this.  I would follow @IamPete1's advice and stick with the stable versions if possible.  It's difficult for us to support the -dev versions on a day-by-day basis, instead we tend to tackle all the various issues as part of the beta testing.

Of course, if one of the other core devs has a chance to look into this, that's a-ok with me.

ninja-zx11, 2020-09-14:
Thanks @rmackay9  and @IamPete1 .I will stick to the stable version for now.