ArduPilot/ardupilot #15325: AP_OSD Pitch Horizon Stay on screen all the time.

devimauz, 2020-09-13:
Currently as I see, horizon works only +-20deg.
For acrobatic this is really bad because when horizon go off the screen you dont know the roll/pitch.
For test i simply modified float pitch = -ahrs.pitch/5; So divided the pitch by 5.
So the horizon stays always on screen and you know roll even going straight up.
In configuration should be good something to switch to acrobatic horizon.

kd0aij, 2020-09-13:
Isn't the horizon line supposed to be drawn *on* the horizon?

devimauz, 2020-09-13:
but horizon line disappear if pitch more than +-20deg

kd0aij, 2020-09-13:
the +/- 20 degree limit is by design, it is hardwired in the display code
I guess people flying acro go by the actual horizon, not the artificial one

kd0aij, 2020-09-13:
perhaps @Hwurzburg or another experienced FPV pilot will comment on this

devimauz, 2020-09-13:
I know 20degree is the limit but if you divide it by 5 then you have full range artifical horizon.
Let's see what other people says.

kd0aij, 2020-09-13:
also keep in mind that euler roll is useless near +/- 90 degrees pitch, so you definitely want to disable the artificial horizon then

vierfuffzig, 2020-09-13:
imho this is what you would commonly use a pitch-ladder type horizon for, which most likely is not a viable option on a charset-based osd, in contrast to a fully graphical overlay type.
breaking general pitch alignment in favour of that corner case (roll display at extreme pitch angles) imho won‘t add much overall benefit.
using the existing individual roll and pitch panels might be an option to display angles beyond the single-bar horizon‘s limitations.