ArduPilot/ardupilot #15326: AP_Arming: remove arming checks 'all'

IamPete1, 2020-09-13:
I have been pondering this, but suspect it will be controversial, so I thought I would open a issue before going to the effort of a PR.

Having the 'all' arming check option encourages users to switch them all off. If instead we were to move to the default being 0xFFFFFFFF people would have to switch them off individually so its more likely that they will keep most. In particular I'm thinking of the mission planner bitmask helper.

[image]

I can see this might be annoying for those without the benefit of the helper. This also makes it more effort to switch them all on again if they were off.

amilcarlucas, 2020-09-13:
I agree with changing the default

rmackay9, 2020-09-13:
.. it's not a bad idea..

peterbarker, 2020-09-14:
So someone flicks a bit off (and unchecks "all").

Saves the parameter to the vehicle.

We allocate a new arming check, "User is happy".

That new option will not be active on that user's vehicle - and will silently be skipped.

These bits should have been options to skip arming checks, not enable them - but given it moved from a boolean to a bitmask....

WickedShell, 2020-09-14:
The other relevant point is `0xFFFFFFFF` doesn't work with the existing float transport system. We can pick the highest one we can actually transport, but that's about it. (And we'd definitely have to remove the `@Values` stuff for sure).

The best solution as @peterbarker pointed out is to swap to making it a skip, rather then an enable, but that's probably fairly breaking everywhere.

rmackay9, 2020-09-14:
I'll add my personal opinion that I think "skip" and "ignore" are not great in general.  I think the double negative just makes it harder for users to understand.  It's a mute point though anyway as others have said above.

IamPete1, 2026-09-27:
Closed by https://github.com/ArduPilot/ardupilot/pull/31568

closed 2026-09-27 by IamPete1