ArduPilot/ardupilot #15329: APM: PreArm: GPS is not healthy, when running Airsim SITL with cygwin

Maxilos, 2020-09-13:
This is a weird problem. Here's how it goes: I've succesfully used WSL before. The way I launched SITL was by running the following command (in ubuntu console)
/home/aaaa/ardupilot/build/sitl/bin/arducopter -S -I0 --home 55.19819900,37.13073700,583,360 --model airsim-copter --defaults /home/aaaa/ardupilot/Tools/autotest/default_params/airsim.parm 
and then running MAVproxy in separate cmd.
This works perfectly fine, then I decided to use cygwin, since I wanted to be able to move the whole SITL installation to other PCs easily.
So, I installed cygwin, did all the steps, and used the same method to launch the SITL, and I got "APM: PreArm: GPS is not healthy" error. I triple checked everything, the setup seems to be equal to the WSL one, and yet, cygwin version just keeps getting this error, no matter what I do. I use the same AirSim project for both, so the problem doesn't seem to be with it. 
Again, the command to launch SITL is basically the same.
I also used custom airsim.parm for both wsl and cygwin.
I'm out of ideas, would really appreciate, if someone could help me out.

djdylex, 2021-01-05:
Was wondering, how can i clone rc2? Not super experienced with git but i don't see that available as a tag. Im getting similar issues with the latest master, when i disable flight checks my copter flights all over the place.

fabioandraderj, 2021-06-08:
I have the same issue. I was using the last release (V4.1.0-dev) and getting the GPS error, then I changed to V4.0.7 and now the quadcopter is behaving "like a drunkard" as @Maxilos said. Do you guys know where I can get a version that works? (maybe the 4.0.4-rc2?)

fabioandraderj, 2021-06-09:
Hi @djdylex , you can clone the repository at the 4.0.4-rc2 version by using the following commands:

git clone -n https://github.com/ArduPilot/ardupilot.git
cd ardupilot
git checkout 93f0c46d7aaf9f1bf14d64d753475fa8733063f4

IamPete1, 2021-08-16:
Is this still a issue on master? @rajat2004 maybe you have tested recently?

fabioandraderj, 2021-08-17:
I was able to make it work fine checking out commit #7292ee8ec6ebcec31acfd0e0849cf165668f4ac2

IamPete1, 2021-08-17:
I think we can close then, Thanks!

closed 2021-08-17 by IamPete1

LightCannon, 2022-02-08:
Hello
@Maxilos @djdylex @professorfabioandrade 
Is there a step-by-step guide to solve this issue? I am experiencing it in the Copter branch. Also, I tried with 4.0.4-rc2 and facing other problems.
Thanks in advance.

professorfabioandrade, 2022-02-10:
Hi @LightCannon, I am able to use AirSim and Ardupilot.

I am using ArduCopter 4.1.0-rc3 and AirSim 1.5.0.

Then I can arm motor, change mode to GUIDED and take off.

If I try to arm in GUIDED mode, I get the GPS error. But if I arm in STABILIZE mode and then change mode to GUIDED afterwards, it works.