ArduPilot/ardupilot #15341: EKF3: Plane: bad Yaw estimate

IamPete1, 2020-09-14:
This is a issue that had been in master for at least 15 days and was not there in Feb. The issue seems to only appear on Plane, and possibly Quadplanes and/or Tailsitters in particular, given the lack of any other reports.

This is a example of EKF3, EKF2 and DCM yaw estimate.
[image]

Roll is much closer to what I would expect (but by no means good)

[image]

Back to back testing has been done with older Master and Current master, showing this behavior is new. Tests have also been done on Copter master using EKF3 but have failed to reproduce the issue.

I have lots of logs of this, with various combinations of params, copter and older master and two different vehicles. I can share on request.

@bugobliterator was kind enough to run some logs through Replay, where the issue is not seen.

I have tryed to reproduce this in SITL but, so far, have not had any success.

kd0aij, 2020-09-14:
Could you post a pointer to some logs?

IamPete1, 2020-09-15:
This was disused on the dev call, the conclusion of this is that there is a issue with the EKF3 initialisation code when run nose up. This is the expected configuration of a tail-sitter. Hence why this was repeatable for this vehicle and not seen elsewhere. Just being off vertical by 5 or 10 deg should be enough to allow it to init correctly. The longer term solution to is to change the initialisation function.

IamPete1, 2021-02-10:
This was fixed by https://github.com/ArduPilot/ardupilot/pull/15351

closed 2021-02-10 by IamPete1