ArduPilot/ardupilot #15345: Save Waypoint for PLANE

RMFPV, 2020-09-15:
## Feature request

Please add Auxiliary Function Save Waypoint for Plane too.

**Describe the solution you'd like**
It's will be very useful to record a route even not that precise, but it's possible to correct it latter manually if needed. Still much more convenient than doing it from scratch.

**Platform**
[+] Plane

IamPete1, 2020-09-15:
You can do this with a lua script

RMFPV, 2020-09-15:
Can you please advise, is there any ready to use script?
I'am not able to write scripts.

It's also nice to have this functionality with Plane by default, as it was already added to Copter and Rover.

IamPete1, 2020-09-16:
This is a example script messing with missions, https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/mission-edit-demo.lua

Would need a little work to trigger on switch.

RMFPV, 2020-09-16:
Great! Thank you very much!
I'll try to test it.

IamPete1, 2020-09-16:
Just to be clear, I would not recommend flying with it. But it works great in SITL, and provides a example of how to add mission items. There are other examples for reading switches.

RMFPV, 2020-09-16:
What can be the problem?