ArduPilot/ardupilot #15346: EKF failsafe change mode to LAND when FS_EKF_ACTION = 2 with RC failsafe

hamzzi, 2020-09-15:
## Bug report

recently I testing failsafe logic in arducopter SITL.
I found something suspicious action in EKF failsafe action

In [EKF failsafe action doc](https://ardupilot.org/copter/docs/parameters.html#fs-ekf-action), EKF failsafe change mode to ALT_HOLD with `FS_EKF_ACTION = 2`.
It was working perfectly when I triggered only EKF Fail.
But when I trigger EKF fail after RC fail mode changed to LAND, not to ALT_HOLD.

Here is my test configuration.
```
FS_THR_ENABLE: 1
FS_EKF_ACTION: 2
```

1. Test 1
trigger only EKF failsafe
```
Guided -> [EKF failsafe] -> ALT_HOLD
```

2. Test 2
trigger RC failsafe and EKF failsafe
```
Guided -> [RC failsafe] -> RTL -> [EKF failsafe] -> LAND
```

There is no mention about RC fail affect to EKF failsafe action in the docs.
I tested change mode to ALT_HOLD and succeeded in the RC fail.

Please check this and tell me if i'm wrong.

**Version**
master (c722367c6c48a01260e7bf6af4edc0e513af5132)

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad Copter

**Hardware type**
SITL

**Logs**
[log.tar.gz](https://github.com/ArduPilot/ardupilot/files/5223121/log.tar.gz)

tatsuy, 2020-09-20:
It seems that it changes the mode to LAND intentionally with RC fail safe.
https://github.com/ArduPilot/ardupilot/blob/c722367c6c48a01260e7bf6af4edc0e513af5132/ArduCopter/ekf_check.cpp#L171
If you can not operate using your transmitter, switching to Alt Hold mode is a high risk because it will be possible to fly away by the wind etc. and lost.
So I think LAND is a good choice.

Should we add it to the documentation?
https://ardupilot.org/copter/docs/ekf-inav-failsafe.html

hamzzi, 2020-10-22:
Thank you for applying my issue.
This issues was lack of my understanding.
But I think it is good for everyone if we update the documentation :)

closed 2020-10-22 by hamzzi