ArduPilot/ardupilot #15348: Telem 1& 2 not the same.

Oeble1966, 2020-09-15:
After installing 4 TF-Lunas with high data rate i noticed difference in Serial 1 & 2. 

Serial 2 is not able to handle high data rate as is serial 1 not able to do this. But Serial 1 reacts to the SR1(Hz) settings and Serial 2 is not reacting to the SR2 settings.
I can use Serial 1 for the Lunas @ 2Hz - > outputs 400bps.
Serial 2 however has a data rate of 10Hz ?  standard , which gives 2000bps of data for the distance sensors ( not changeable trough settings = not responding to SR2 settings ) and therefore is this port not usable for (in my case ) Minimosd.

2Hz = 400 bps on port serial 1
1Hz =200 bps on port serial 1
Guess 10Hz is the 2000 bps what i see on Serial port 2

 _Please describe the problem_
See this link and the follow up link below.

https://discuss.ardupilot.org/t/new-lidar-sensor-benewake-tf02-pro-tfmini-s-tf-luna-for-drones-uav/58614/17

**Version**
Ardupilot / copter 4.04 RC4 Beta  and 4.03 

**Platform**
[ x] All

**Airframe type**
All as it is a Pixhawk BUG.

**Hardware type**
Pixhawk 2.4.8.

see also discussion below, 
https://discuss.ardupilot.org/t/not-quite-solved-tf-luna-or-any-rangefinder-creating-bad-connection-between-air-and-ground-units-mavlink/61332/2

magicrub, 2020-09-15:
Those serial ports should be the same. Is your flow control set to auto? Try disabling it. with BRD_SER1_RTSCTS = 0 and BRD_SER2_RTSCTS = 0.

You talk about SR1 and SR2, but don't say which ones you're setting.

Oeble1966, 2020-09-15:
Hi thanks for the reply ,
Yes BRD_SER 1&2 where Auto , now both are set to 0.

In Telem 2 the Sensors SR2_RAW_SENS set the Lidars Hz.
This works in full list and MP planner screen.
Loading of params also gets slow when set to 10Hz ( same as on Telem 1 )

In Telem 1 nothing happens. ( i see the telemetry stream getting blocked and almost not refreshing ) 
This was what i also noticed @ first mounting of the Luna's on the 915<hz sik radio link.
Looks like its always on 10Hz.

magicrub, 2020-09-15:
If you suspect ArduPilot is to blame, please remove external links to reduce the scope.

Oeble1966, 2020-09-18:
Hi ,

So did some research.
What i need ( and probably others )  is to change the Hz of the TF-Luna's .
This is possible on the luna's.
However as we are using the "AP_RangeFinder_Benewake_TFMiniPlus" as driver.
It seems the Luna is set to default settings on every reboot of the Pixhawk ?. 

Am i correct here ?.

closed 2020-10-01 by Oeble1966