ArduPilot/ardupilot #15353: EKF failsafe dose not change to LAND mode when ALT_HOLD

hamzzi, 2020-09-16:
## Bug report

**Issue details**

Recently I testing failsafe logic in arducopter SITL.

FS_EKF_ACTION = 1 change mode to LAND when EKF failsafe is occurred.
It is working on other mode like GUIDED mode, but when current mode is ALT_HOLD, mode is not changed to LAND.

Here is mode flow.
```
Takeoff -> ALT_HOLD -> [EKF failsafe] -> ALT_HOLD (not changed)
```

I think it should be
```
Takeoff -> ALT_HOLD -> [EKF failsafe] -> LAND
```

**Version**
master (c722367c6c48a01260e7bf6af4edc0e513af5132)

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad Copter

**Hardware type**
SITL

**Logs**
[log.tar.gz](https://github.com/ArduPilot/ardupilot/files/5229840/log.tar.gz)

rmackay9, 2020-09-16:
There shouldn't be any need to switch into Land mode on an EKF failsafe because AltHold's horizontal position is controlled by the pilot.  In general the EKF failsafe is only for loss of the horizontal position estimate.  It's quite rare for a vehicle to lose control of it's vertical access but for that we have the vibration compensation.

So I think there's no issue here and we can close this issue.

hamzzi, 2020-09-16:
Thank you for answer.
I found your answer in the doc.
```
In manual flight modes that do not require GPS (i.e. Stabilize, Acro, AltHold) nothing further will happen but the pilot will be unable to switch into autonomous flight modes (Loiter, PosHold, RTL, Guided, Auto) until the failure clears
```
Doc said if current mode dose not require GPS, it ignores EKF failsafe.
But FS_EKF_ACTION = 3 which means LAND even in Stabilize works on ALT_HOLD mode.
Stabilize means EKF failsafe is applied even if it is no GPS mode?

rmackay9, 2020-09-16:
@hamzzi, ah, yes, option 3 should be reworded to replace "Stabilize" with "Manual" or maybe a list of all modes that don't require GPS including AltHold.  Put another way, if FS_EKF_ACTION = 3 then I think the failsafe should trigger in all modes including AltHold.

hamzzi, 2020-09-16:
@rmackay9
Thank you for kind anwser :)

closed 2020-09-16 by hamzzi