ArduPilot/ardupilot #15364: Questions about update_throttle_mix()

lukezhqin, 2020-09-17:
## Bug report
I found some questions recently in update_throttle_mix() funcion.
**Issue details**
1, in land_detector.cpp Line 154, `channel_throttle->get_control_in() <= 0`, I checked that channel_throttle return between 0~1000, so what do you mean by "<=" 0? I don't sure I got the point you meight mean here.

2, also in land_detector.cpp  Line 152, I thought the code `flightmode->has_manual_throttle()` means we are in manual command such as poshold, althold mode according only to the following comments. Then I surprisingly found my mistake that it only means two mode "acro" "stabilize". I want to understand this kind of design in this `update_throttle_mix()` function, could you guys help understand it better?

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-09-17:
@lukezhqin,

- RC_Channel::get_control_in() returns an int16_t so it is best to check for <= 0 instead of just 0.
- Mode::has_manual_throttle() returns true for modes where the pilot directly controls the throttle which includes Stabilize and Acro mode.  It's possible we could make this easier to understand by adding comments above each of the virtual functions in the Mode class.
- the purpose of update_throttle_mix is to set the prioritisation of attitude control vs altitude control.  In modes where the pilot directly controls the throttle we have a specific parameter that sets the prioritisation, MOT_THR_MIX_MAN.  Manual throttle modes are the easiest to deal with because we know a pilot is in control. for the autopilot controlled modes it's more complicated because we need to worry about getting the mix right so that the vehicle stays upright but we also need to worry about a wobbly copter's rate controller trying to steal all the motor bandwidth and causing the altitude controller not to work properly (which can lead to fast descents or fast climbs)

closed 2020-09-17 by rmackay9

lukezhqin, 2020-09-18:
Thanks very much for your help !!