ArduPilot/ardupilot #15397: EKF2 and EKF3 significantly unhappy in QuadPlane GyroFFT test

peterbarker, 2020-09-22:
## Bug report

**Issue details**

The GyroFFT sporadically fails in autotest - the quadplane fails to disarm; the physics is failing, I think, as we end up 15m underground.

The console output shows EKF2 and EKF3 deeply unhappy about their magnetometers, and the last logfile for the test has this:

[image]

and

[image]

I've uploaded that log below, 'though it is trivial to reproduce it.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadPlane

**Hardware type**
SITL

**Logs**

Can be generated with `./Tools/autotest/autotest.py --gdb --debug build.Plane test.QuadPlane.GyroFFT`

.TXT can be stripped off this file (.TXT added to I can upload it):

[00000003.BIN.TXT](https://github.com/ArduPilot/ardupilot/files/5259018/00000003.BIN.TXT)

tridge, 2020-09-22:
this happens because SIM_VIB produces sensor data that is not kinematically consistent. The implementation of SIM_VIBE needs to be moved into SITL and needs to change the model before sensor smoothing is applied

tridge, 2020-09-22:
@andyp1per do you want to look at this? We need kinematically consistent sensors

peterbarker, 2020-09-22:
On Mon, 21 Sep 2020, Andrew Tridgell wrote:

> this happens because SIM_VIB produces sensor data that is not kinematically consistent. The implementation of SIM_VIBE needs to be moved into SITL and
> needs to change the model before sensor smoothing is applied

Should we disable this test until that is done?

I note there are quite a few other tests which use this.  I guess this 
might be something to look out for in other tests failures.

tridge, 2020-09-22:
maybe run with AHRS_EKF_TYPE=10 for now?

peterbarker, 2020-09-22:
On Mon, 21 Sep 2020, Andrew Tridgell wrote:

> maybe run with AHRS_EKF_TYPE=10 for now?

It *is* running with EKF type 10 :-)

Which I guess does raise the question on why things go so wonky with the 
q-landing.

andyp1per, 2020-09-22:
@tridge I guess I don't understand what change is being asked for - can you give me an example to get my head around?

andyp1per, 2020-09-24:
I don't see where the mag field is updated with the sensor physics - and in reality I see the same level of innovation on real aircraft with bad vibrations, so I'm not convinced there is a problem here. Also there is only gyro noise added - no accel noise. SIM_RND is used everywhere and has been for a long time - you can't just turn it off. Surely the goal here is to make the test more reliable - I'm pretty sure I can do that as the mission where this shows up has a notch to squash the noise, so in reality where it matters it should be possible to not disturb the physics too much.