ArduPilot/ardupilot #15428: GPS: Extend ublox RAW/RAWX logging to improve PPK capabilities

Yups1337, 2020-09-25:
**Feature request**

As RTK GPS corrections need a realtime link to the base-station, this is not the best way to go when the exact flight path of the UAS isn't critical. For typical mapping applications post-processing (PPK) is the best and most reliable solution. To do this, we need to log two extra messages from the ublox receivers:

**UBX-RXM-SFRBX**
Broadcast navigation data subframe

**UBX-TIM-TM2**
Time mark data

The TIM-TIM2 message is used for camera feedback, each time EXTINT1 (or 2) is pulled low, the exact GPS time is recorded.
Using RTKLib emlid for PPK gives an interpolated positions file with all events.

Nice tutorial by **rtklibexplorer**
https://rtklibexplorer.wordpress.com/2018/10/26/event-logging-with-rtklib-and-the-u-blox-m8t-receiver/

This works with ublox M8T, M8P, F9T and F9P.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

Best regards form Germany
Yannick

Naterater, 2020-09-26:
I 100% support any development toward this. This issue is currently causing us to use external hardware for PPK logging.

Naterater, 2020-10-06:
This would also solve https://github.com/ArduPilot/ardupilot/issues/10369#issue-404940171 in another way.

Naterater, 2021-10-01:
@tridge, Seeing the prevalence of the Here3 being extremely low-cost PPK capable since it has the interrupt pin on the board, I'd REALLY like for this to be supported.