ArduPilot/ardupilot #15452: Copter: Internal errors 0x4000 I:211 spi_fail

Notorious811, 2020-09-28:
## Bug report

**Issue details**
Was flying my 5" quadcopter on omnibusf4pro, after few flights suddenly got this error when it was on the ground, i saw yellow LED flashing on GPS, when i connected to mission planner it was showing this error **PreArm: Internal errors 0x4000 I:211 ,spi_fail**, then i removed all the arming checks just for checking then GPS LED started blinking back to green, but after removing all arming checks this message was still popping up continuously on message console and on HUD, haven't taken any flight after this issue as it seems risky, just rebooted the FC and this error message didn't not show up anymore

One thing which i observed with this particular drone is whenever i power up this drone, after arming it first time only there is GPS glitch most of the time. 

Second important observation after looking at the log was that logs stopped building when this error started occurring, i am not very sure about it as i can see that plots not building in the graphs but mode change still can be seen. This suggest me that this issue probably related with SD card which is working on SPI too.

**Version**
ArduCopter v4.1.0-dev (5e6a4b29)

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad - X

**Hardware type**
OmnibusF4pro

**Logs**
https://drive.google.com/file/d/1yjtTCUsuYlYWZaQRlNP-XkUaztlo_IGf/view?usp=sharing

Please ignore the radio failsafe in log, remote was off during that times

[image]

[image]

giacomo892, 2020-09-28:
some discussion going on here #12773 too

peterbarker, 2020-10-05:
I've had a KakuteF7 running for ~21 days on the desk behind me.  this is a bare board - USB is plugged in and nothing else - not even an SD card.

Sometime in that last 20-odd days, the IMU had an issue:

```
STABILIZE> param fetch
STABILIZE> Requested parameter list (ftp)
APM: ArduCopter V4.1.0-dev (0e1631ce)
APM: ChibiOS: 331fe75d
APM: KakuteF7 00200025 3538510B 31373730
APM: RCOut: PWM:1-6
APM: IMU0: fast sampling enabled 8.0kHz/2.0kHz
APM: Frame: UNKNOWN
Received 1041 parameters (ftp)
Saved 1041 parameters to mav.parm
arm throttle
STABILIZE> APM: PreArm: Check firmware or FRAME_CLASS
APM: PreArm: Throttle below Failsafe
APM: PreArm: 3D Accel calibration needed
APM: PreArm: EKF attitude is bad
APM: PreArm: Compass not healthy
APM: PreArm: AHRS not healthy
APM: PreArm: Internal errors 0x1004000 l:212 spi_fail,imu_reset
Got COMMAND_ACK: COMPONENT_ARM_DISARM: FAILED
```

IamPete1, 2022-08-05:
Any updates to this?

rushalexander, 2022-09-04:
Any updates to this?

andyp1per, 2022-09-04:
There have been a number of fixes for the omnibusf4pro in ArduPilot 4.2, including refreshes to ChibiOS. If the issue does still exist I would not expect it to be the same.

TretUliy2, 2024-11-09:
Having error in Arduplane 4.5.7 FS Matek F405-WTE looks almost the same "PreArm: Internal errors 0x4000 l:215 spi_fail"