ArduPilot/ardupilot #15455: Please Add support for MMC5883MA compass

ninja-zx11, 2020-09-28:
## Feature request

Please add support for MMC5883MA magnetometer.This magnetometer is even less noisy than HMC5883L.
Driver is present on pixhawk:https://github.com/traxxas/PX4Firmware/commit/89fe5e7bdf7a91a5bed7a6c88f35a5b56dcbff98#diff-8989715f5461b08efdb89011d2e99b5c

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

[image]

auturgy, 2020-10-20:
what autopilot uses this sensor?

pepak, 2021-04-07:
MMC5883MA driver has been added: [#16976](https://github.com/ArduPilot/ardupilot/pull/16976)

ninja-zx11, 2021-04-07:
Thanks pepak. That's awesome. I will test my board soon.

closed 2021-04-07 by ninja-zx11