ArduPilot/ardupilot #15461: Emit log message and statustext

peterbarker, 2020-09-29:
## Feature request

**Is your feature request related to a problem? Please describe.**
When we raise an internal error it may be quite some time before we get something in the dataflash log or in the mavlink stream.  This makes it harder to correlate internal errors with codepaths.

**Describe the solution you'd like**
Emit a statustext and/or log message when we detect an internal error has been raised

**Describe alternatives you've considered**
Reading tea-leaves.

**Platform**

- [X] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover
- [ ] Submarine

**Additional context**

It might be that this can't be done directly when raising the error - perhaps polling the flags at the main loop rate might be required.  If nothing else, InternalError doesn't currently include libraries required to do that, and almost certainly doesn't want them.

DhairyaBahl, 2020-10-05:
I would like to contribute to this issue. Kindly assign it to me.

Thanks

peterbarker, 2020-10-05:
@DhairyaBahl I can't assign it from a technical github perspective - but people will see these messages and act accordingly.  Good luck!

Sobhit25, 2020-11-05:
I want to contribute to Ardupilot, as my skillset match its requirements.
Could you please mentor me and guide me through the code base.

peterbarker, 2020-11-17:
@DhairyaBahl any movement on this?

@Sobhit25 check out our ArduPilot development wiki @ dev.ardupilot.org

NinaPacifier, 2020-12-12:
@peterbarker May I work on this issue???

peterbarker, 2020-12-12:
On Sat, 12 Dec 2020, Nishant Nandan wrote:

> @peterbarker May I work on this issue???

Sure!

Sobhit25, 2021-01-20:
Goodmorning @peterbarker ,
I did set up an environment on linux. SITL and MavProxy are working fine.
Can simulation be used to fix the issue, or I need to have hardware?
Could you tell me Which files to follow along for understanding the internal error code.

peterbarker, 2021-01-21:
@Sobhit25 yes, this issue can be resolved in SITL.

The goal is to announce we've had a problem as soon as we can.

We can't do it immediately the error is raised (https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_InternalError/AP_InternalError.cpp#L11) as we may be in an interrupt context and we can't spend time doing random things in there.

What we can do is poll the internal error state in the main loop (perhaps in here: https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_Vehicle/AP_Vehicle.cpp#L160).

If the internal-error mask has changed since we last saw it then we should emit a message to the log (probably a new logger event  (https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_Logger/AP_Logger.h#L163) and a statustext to the GCS.

Learning how to use GDB might help: https://ardupilot.org/dev/docs/debugging-with-gdb.html

PR #16763 (2021-08-31): AP_Vehicle: log internal error and report bitmask to GCS
Attempt at resolving https://github.com/ArduPilot/ardupilot/issues/15461 

I have an issue using the sendtext, local build sitl require %x but other board need %lx ? 

I also do not know best way to test, if someone could please advise.

Fixes #15461
closed 2021-08-31 by peterbarker