ArduPilot/ardupilot #15465: Copter: Internal errors 0x400 I:269 cnstring_nan

Notorious811, 2020-09-29:
## Bug report

**Issue details**
When doing compass/motor calibration on my 7" pixhawk1 and 5" omnibusf4pro quadcopters using just throttle, then i am getting this **PreArm: Internal errors 0x400 I:269 ,cnstring_nan** continuously on the message console and mission planner HUD, it can take around or more than 30 seconds before this error message starts popping up after clicking finish when doing compass/motor calibration. **This issue also getting reproduce in my case every time**. On rebooting the board, this message does not show up anymore.

**Version**
ArduCopter v4.1.0-dev (5e6a4b29)

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad-X

**Hardware type**
Pixhawk 2.4.8

**Logs**
https://drive.google.com/file/d/1KMAyrUL20iRHtqjoMe3AFAyQEv-R2M52/view?usp=sharing

[image]

peterbarker, 2020-10-13:
@WickedShell has some patches which might help us track this down.

WickedShell, 2020-10-13:
The main commit of relevance is 174549501a6afbce168a3b5e916ea4f78f57173c which gives us the line number the constrain nan was called from, which can help a lot in investigating it. We may not want to do this on the 1MB boards, but I think it's a clear win on the 2MB boards and I'll probably PR it in the near future after I've had a chance to test the impact on the 1MB boards.

peterbarker, 2020-11-17:
@Notorious811 are you able to run again with @WickedShell 's patches?  Master will work on >1MB boards.

Notorious811, 2020-11-17:
> @Notorious811 are you able to run again with @WickedShell 's patches? 

Actually nowadays i am working mostly with arducopter stable version v4.0.4, its been a while since i used master version on actual quadcopter, this PR #15662 is next in my to-do list, i will test it soon with master after quickly dealing with RTC based time synchronisation stuff, will try to update in a day or 2

Notorious811, 2020-11-21:
@WickedShell @peterbarker this is the log running arducopter v4.1.0-dev with this patch #15662 on Pixhawk1, just bench tested atm, did compass/motor calibration using throttle only, but this issue also occur after doing this calibration with current sensor 
https://drive.google.com/file/d/1G0rUMUUgpGJ7tOT_YiLC3KWpcIVeMmIo/view?usp=sharing

Notorious811, 2020-11-25:
@peterbarker did the log helped in finding the source of this internal error ?

peterbarker, 2020-11-26:
Thanks @Notorious811 , really handy!

@priseborough this appears to be something in the EKFGSF.

The bug seems to be on this line:
```
            AHRS[mdl_idx].gyro_bias[i] = constrain_float(AHRS[mdl_idx].gyro_bias[i], -gyro_bias_limit, gyro_bias_limit);
```
(https://github.com/ardupilot/ardupilot/blob/wip%2Fwatering-system/libraries/AP_NavEKF/EKFGSF_yaw.cpp#L225)

So, one scenario as to how this could happen:

Looks like we don't have protection about the accel vector coming in very low.  If `ahrs_accel_norm` is zero or a very small number then we may end up with `tilt_error_gyro_correction` being Inf (or NaN) (on line 214), that Inf/Nan propagated into `AHRS[mdl_idx].gyro_bias` and then caught when we try to constrain that value.

Given the filter on `ahrs_accel`, that would seem to infer the first call to the EKFGSF's update call would need to be passed a delVel of 0.0f....

peterbarker, 2020-11-26:
@Notorious811 I don't suppose I could ask you to make the same sort of log but enable `LOG_REPLAY` and `LOG_DISARMED`, could I?  Please make sure you have an excellent SD card or a large `LOG_FILE_BUFSIZE` to reduce the number of lost packets.

Notorious811, 2020-11-26:
@peterbarker this is the log with LOG_REPLAY=1, LOG_DISARMED=1, LOG_FILE_BUFSIZE=64 and a better SD card
https://drive.google.com/file/d/1GHETwCll1ou7_h-SX5gj8PsMFr4NrLtk/view?usp=sharing

please do let me know whenever you want to test something especially for bugs/issues with master, i also want to ensure that everything working as supposed to even if they are some trivial bugs/issues.

peterbarker, 2020-11-26:
@Notorious811 could you do the same thing - but reboot the vehicle after setting those parameters, please?

Turns out our new Replay really doesn't like coming in part-way through the data :-)

Notorious811, 2020-11-27:
@peterbarker I think I did rebooted the vehicle after setting the parameters and before starting compass/motor calibration, but I can try again, will send the log in 3-4 hours

Notorious811, 2020-11-27:
@peterbarker its getting weird now, i am not able to reproduce this issue now, have done compass/motor calibration several times, but not getting this internal error anymore, this is one of the log
https://drive.google.com/file/d/1ZP3tFYLf9-D3oRkD1vVbvHuz7N1FqbO9/view?usp=sharing

peterbarker, 2020-12-07:
@Notorious811 still no luck reproducing?  If we're lucky then its been accidentally fixed.  If we're unlucky we've moved the problem....

Notorious811, 2020-12-07:
@peterbarker haven't tested in a while, think we should move forward, this particular issue does not look anything serious, and occasionally reproduce during Compass/motor calibration step only and goes away after reboot, should I close this issue ?

peterbarker, 2021-01-19:
@Notorious811 please reopen this issue if you see it again.  Thanks for reporting.

closed 2021-01-19 by peterbarker