ArduPilot/ardupilot #15477: UAVCAN node id falsely identified as duplicate id

flytrexguy, 2020-10-01:
## Bug report

**Issue details**

The map of node's unique hardware id to UAVCAN node id is stored in persistent storage. So when connecting a different device with the same node id (after disconnecting the first device and rebooting AP of course) it will report it as duplicate id.

How to recreate the bug:
Connect a UAVCAN device with a static node id. Then reboot AP and connect a different device (with a different hw id) with the same static node id. 

**Version**
Copter-4.0.4

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
hex

**Hardware type**
mRoControlZeroF7

Vkd4U2NtVkZOVg, 2020-10-31:
Have the same issue with CUAV X7 Pro and UAVCAN pmu, I have two same pmu with the same id, one of them is broken, so I replace with the other one, but the PreArm: UC: Duplicate Node com.hex.here../xx pops up. So I have to disable the pmu buy now. any idea how to solve this?

JacksonUAS, 2023-04-30:
Was this ever resolved? Whats the work around? I think ive managed to recreate it :)

tridge, 2023-05-12:
@JacksonUAS I've pinged @SchrodingersGat

David-OConnor, 2023-07-22:
I'm getting this too. Anything I can do to help?

SchrodingersGat, 2023-07-24:
From memory, @JacksonUAS resolved the issue by  clearing the DNA cache

JacksonUAS, 2023-07-25:
Never found a solution, clearing the cache solved it for a bit, then it came back. Went to Piccolo, no issues since.

magicrub, 2023-07-25:
Can people pretty please state what version they are running when reporting a problem.

WickedShell, 2023-07-25:
I've seen and resolved this on 4.0 aircraft with clearing the DNA cache.

> So when connecting a different device with the same node id (after disconnecting the first device and rebooting AP of course) it will report it as duplicate id.

This sounds like it's working correctly? You used a different device with the same ID, that sounds like an error, what have I missed?

JacksonUAS, 2023-07-25:
Connecting the same device, with the same node ID, produces the error as well.

On the test aircraft we were using, ECSC power was separate to avionics. Booting up the ESCs after the FC, and populating them on the bus, with static IDs, generated new instances in the DNA cache....

David-OConnor, 2023-07-25:
Same issue, re same device/same id.