ArduPilot/ardupilot #15480: Engines starts automatically when switched in Q-STABIL without beying armed when using 3D ESC's (PWM initialisation isue)

dramida, 2020-10-01:
## Bug report

**Issue details**

I am using 3D ESC's so 1500 to 2000us is the forward working range and 1500 to 1000 is backward which i don't need.
The isue is repetable, on a fresh Plane install on Pixhawk configure 

I set up:
Q_ENABLE=1
Q_Frame_Class= 10
Q_frame_TYPE 15

Servo3_Function 73
Servo3_Max 2000
Servo3_MIN 1520
Servo3_Trim 1520
Servo4_Function 74
Servo4_Max 2000
Servo4_MIN 1520
Servo3_Trim 1520

When in FBWA, left and right throttle stays as programmed 1520us but when switched in Q-STABIL, even if it is not armed, pwm goes down to 1000 !! This spins motors backwards.

Please refine VTAIL initialisation procedures to take into account defined PWM range.

**Version**
10/1/2020 1:48:56 PM : fmuv2 004F003B 32385114 38383435
10/1/2020 1:48:56 PM : ChibiOS: d4fce84e
10/1/2020 1:48:56 PM : ArduPlane V4.0.6 (036ad450)

**Platform**
PLANE- TAILSITTER

**Airframe type**
two motor tailsitter

**Hardware type**
Pixhawk

**Logs**
no need to log, you may see status parameters CH3out and CH4out

magicrub, 2020-10-01:
yeah, that certainly seems wrong! We'll look into it.

magicrub, 2020-10-01:
I don't know why, but there's a dedicated param that looks like it controls this: Q_THR_MIN_PWM. Please check/test that and report back.

magicrub, 2020-10-01:
@kd0aij could you take a look? I see ```1000``` hardcoded in lots of places in the quad/tailsitter stuff.

kd0aij, 2020-10-01:
It looks like setting Q_THR_MIN_PWM to 1500 should fix this problem.
@dramida  Have you tried that?

kd0aij, 2020-10-01:
@Hwurzburg Does the wiki explain the relationship between SERVOn_MIN and Q_THR_MIN_PWM for motor channels?

Naterater, 2020-10-19:
This should probably be a prearm check that checks the Q_THR_MIN_PWM is within the range of throttle or motor servo ranges.

IamPete1, 2021-08-16:
This should have been fixed for throttle by https://github.com/ArduPilot/ardupilot/pull/16908 I'm faily sure throttle left and right should be OK. @dramida have you tested on 4.1?

IamPete1, 2021-09-06:
@dramida any update?

closed 2021-09-11 by IamPete1