ArduPilot/ardupilot #15488: Copter: In-Flight Watchdog reset, Hard fault

DanS87, 2020-10-02:
Issue details
Autopilot was reset shortly before completing its flight. The Mission was controlled by a companion.

The watchdog error:
[image]

Logs
[before reset](https://drive.google.com/file/d/1qZ_530QpTK0kgDvj7rmp3w4QYCwD_lJc/view?usp=sharing) 
[after reset](https://drive.google.com/file/d/1mIUm4VAYxgJcCe-b0gWDiSY1sMPrjl-C/view?usp=sharing)

Version
ArduCopter 4.0.4

Platform
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
Hexa

Hardware type
mro zero f7

rmackay9, 2020-10-05:
Thanks very much for this report including the logs.  This is pretty serious of course so we are looking into this.

DanS87, 2020-10-07:
We were able to reproduce the issue again:
Logfiles:
[before](https://drive.google.com/file/d/1jbwyZMUuoAq6_4cDWA_G_MWLeGdY3Ni4/view?usp=sharing)
[after](https://drive.google.com/file/d/10PQ9UmG4GdW--K76flKnyg31HyieGXgt/view?usp=sharing)

rmackay9, 2020-10-07:
@DanS87,

Thanks again for the report.  It looks like Tridge may have a fix in this PR: https://github.com/ArduPilot/ardupilot/pull/15525

IamPete1, 2021-08-16:
@DanS87 have you tested https://github.com/ArduPilot/ardupilot/pull/15525 ? (4.1 will include that fix)

IamPete1, 2021-09-06:
@DanS87 any update?

DanS87, 2021-09-08:
@IamPete1 
The issue did not occur again, so pretty sure its fixed

closed 2021-09-08 by IamPete1