ArduPilot/ardupilot #15515: Copter terrain failsafe issues

peterbarker, 2020-10-06:
## Bug report

**Issue details**

Copter has multiple issues with terrain failsafe with a configured but not used rangefinder. And terrain.

I have not fully characterised everything that's going on here.

@mirkix pointed out that a simple mission in SITL produced some bad results.

https://github.com/ArduPilot/ardupilot/pull/15514 contains this mission:
```
0	0	0	16	0.000000	0.000000	0.000000	0.000000	-35.363262	149.165237	584.030029	1
1	0	10	16	1.000000	0.000000	0.000000	0.000000	-35.362487	149.164783	10.000000	1
2	0	3	16	1.000000	0.000000	0.000000	0.000000	-35.361870	149.165747	10.000000	1
```

[image]

When the new autotest is run (`killall gdb xterm ;  ./Tools/autotest/autotest.py --gdb --debug test.Copter.RangeFinderMissionRTL --map`), this is what happens:

[image]

[image]

Note the "RTL" mode?  That's because a terrain failsafe triggers, and we momentarily end up in RTL mode.  We actually enter RTL mode but then trundle on through the "auto wp" code:

It's *much* worse if you don't use guided mode to move around but move away using loiter mode: https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1137 - that call can `set_mode` to RTL on terrain failsafe, which will reset targets etc - but then we continue in auto-mode (I've no idea what switches the mode back to auto....)

[image]

In that image, we think we're going to waypoint one (see the console text), *actually* moving towards home (or, more likely, origin) (see map) - and crashing into the ground part-way there (see rangefinder graph).

As mentioned, haven't fully characterised this.  It feels like potentially several bugs.

In addition to any bugs here, we should probably not allow arming if the mission contains AGL mission items and the user doesn't have a working terrain altitude source.

**Version**
master

**Platform**
- [ ] All
- [ ] AntennaTracker
- [X] Copter
- [ ] Plane
- [ ] Rover
- [ ] Submarine

**Airframe type**
+

**Hardware type**
SITL

**Logs**

mirkix, 2020-11-03:
@rmackay9 @peterbarker 
The reason for not (or only short) switching to RTL in this case is a nested mode_switch call.

Set_mode to auto will be triggered but in set_mode Mode_Auto::init will be called which then try to set RTL with another set_mode call (while the first set_mode is still running) it sets the mode to RTL then all is going back in the Stack Trace and the first set_mode (to auto) will finished and set it back to auto. Because there are a lot of functions calls between the two set_mode calls it is not so easy to fix.

[image]

peterbarker, 2025-06-29:
Tested this again today.  
 - we're no longer running into the ground
 - we no longer have the double-mode-change issue

I think we've just fixed these elsewhere over time.

Closing this one now.  Wish I'd had this autotest committed-and-failing and in the don't-run list.

closed 2025-06-29 by peterbarker