ArduPilot/ardupilot #15531: Please add hardware Matek  F722-STD STM32F722 F7  to use ArduPilot.

fdapport, 2020-10-07:
Could you please implement  the hardware Matek  F722-STD STM32F722 F7  to use ArduPilot. The hardware print is the same of Matek F405-STD  STM32F405 but with more power,, include the possibility to use FrSky SPort telemetry. To use other firmware is not the same that use ArduPilot. We like ArduPilot. Thanks, Franco

IamPete1, 2020-10-07:
The F722 does not have enough flash to fit ArduPilot. We support 1MB boards with a reduced feature set but to cut AP down to 512kB would leave you with something that would be barely usable.

MATEK have recently announced a H743 board that has more power than the F222 and plenty (2MB) of flash to fit all features.

http://www.mateksys.com/?portfolio=h743-slim

fdapport, 2020-10-07:
Great!!!!    I will wait to find a H743 slim board,.  Thanks for your quick replay!! Franco

closed 2020-10-07 by fdapport