ArduPilot/ardupilot #15534: different hwdef.h for python2 and python3

tridge, 2020-10-07:
using python2 vs python3 generates a different hwdef.h for CubeOrange (and likely many other boards). The cause is almost certainly a missing sort in the DMA generator code.

WickedShell, 2020-10-07:
CubeOrange results:
``` diff
% diff hwdef-py2.h hwdef-py3.h           
238c238
< #define HAL_VALIDATE_BOARD (!((spi_check_register("icm20948_ext", INV2REG_WHOAMI, INV2_WHOAMI_ICM20948)))?"$CHECK_IMU1_PRESENT":!((spi_check_register("icm20602_ext", MPUREG_WHOAMI, MPU_WHOAMI_ICM20602)))?"$CHECK_IMU0_PRESENT":!((check_ms5611("ms5611_ext")))?"$CHECK_BARO1_PRESENT":!((spi_check_register("icm20948", INV2REG_WHOAMI, INV2_WHOAMI_ICM20649)))?"$CHECK_IMU2_PRESENT":!((check_ms5611("ms5611")))?"$CHECK_BARO0_PRESENT":nullptr)
---
> #define HAL_VALIDATE_BOARD (!((spi_check_register("icm20602_ext", MPUREG_WHOAMI, MPU_WHOAMI_ICM20602)))?"$CHECK_IMU0_PRESENT":!((spi_check_register("icm20948_ext", INV2REG_WHOAMI, INV2_WHOAMI_ICM20948)))?"$CHECK_IMU1_PRESENT":!((spi_check_register("icm20948", INV2REG_WHOAMI, INV2_WHOAMI_ICM20649)))?"$CHECK_IMU2_PRESENT":!((check_ms5611("ms5611")))?"$CHECK_BARO0_PRESENT":!((check_ms5611("ms5611_ext")))?"$CHECK_BARO1_PRESENT":nullptr)
272c272
< #define STM32_I2C_I2C1_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_I2C_I2C1_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
276c276
< #define STM32_I2C_I2C2_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_I2C_I2C2_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
284c284
< #define STM32_SPI_SPI2_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_SPI_SPI2_TX_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
290c290
< #define STM32_TIM_TIM1_UP_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_TIM_TIM1_UP_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
292c292
< #define STM32_TIM_TIM4_UP_DMA_STREAM   STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_TIM_TIM4_UP_DMA_STREAM   STM32_DMA_STREAM_ID(2, 4)
294c294
< #define STM32_UART_UART4_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 4)
---
> #define STM32_UART_UART4_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
298c298
< #define STM32_UART_UART7_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_UART_UART7_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
304c304
< #define STM32_UART_USART2_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_UART_USART2_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
308c308
< #define STM32_UART_USART3_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared UART7_TX,TIM4_UP,I2C2_TX,SPI2_TX,USART3_TX,USART2_TX,I2C1_TX,TIM1_UP
---
> #define STM32_UART_USART3_TX_DMA_STREAM STM32_DMA_STREAM_ID(2, 3) // shared TIM1_UP,SPI2_TX,UART4_TX,I2C1_TX,I2C2_TX,USART2_TX,USART3_TX,UART7_TX
```

I noticed this because I was having stability problems with RTK injection when building with python3

WickedShell, 2021-04-13:
@tridge any chance of this getting some attention before shipping 4.1? It's getting harder to match the build servers these days as more things are moving off python2/dropping it, and the fact that your build env significantly changes DMA mappings is not good...

PR #15535 (2023-04-25): HAL_ChibiOS: fixed hwdef DMA resolver dependency on python2/python3
we need to use sorted keys in dictionaries to ensure we end up with
the same DMA mapping when using py2 and py3

fixes #15534

ping @WickedShell
commit 5024261e (2023-04-25): HAL_ChibiOS: fixed hwdef DMA resolver dependency on python2/python3

we need to use sorted keys in dictionaries to ensure we end up with
the same DMA mapping when using py2 and py3

fixes #15534
closed 2023-04-25 by tridge