ArduPilot/ardupilot #15545: Copter: Autotrim evolution

robustini, 2020-10-09:
It would be very nice if the "Autotrim" functionality in the Copter was automated in relation to the Loiter and Position Hold level, obviously always activated via a switch to save these trims and operation to be carried out without wind.
Currently you have to trim manually, but there shouldn't be this need since when the copter stays in position it actually knows very well how the trims should be set.
It could also be very useful in "non-vectored tailsitters", where often the aileron trims in Copter are different from that of Plane, currently it is a parameter to be set manually, but if it were automated it would be great.
A small improvement that could help a lot.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

robustini, 2020-10-09:
LOL, it has been working like this for some time, i'm sorry!

robustini, 2020-10-09:
But I don't think that's possible with non-vectored tailsitter, it would be useful.

closed 2020-10-09 by robustini

peterbarker, 2020-10-09:
On Fri, 9 Oct 2020, Marco Robustini wrote:

> Currently you have to trim manually, but there shouldn't be this need
> since when the copter stays in position it actually knows very well how
> the trims should be set.

Only if there are no external influences on the vehicle - wind being the 
most obvious.