ArduPilot/ardupilot #15547: Problem with Ardupilot initialisation on Linux - bbbmini

juvinski, 2020-10-09:
## Bug report
Trying to use ardupilot software with Linux -kernel 4.14 and 4.19 - I'm having this problem:

1) Initializing ardupilot - plane or copter - the INS is not detected:
root@beaglebone:~# ./ardupilot/build/bbbmini/bin/arduplane -A udp:192.168.7.1:14550
MS5611 found on bus 2 address 0x01
MS4525: no sensor found
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver

2) If I run the INS_generic sample, the INS is detected:
root@beaglebone:~# ./ardupilot/build/bbbmini/examples/INS_generic 
AP_InertialSensor startup...
MPU[0]: enabled fast sampling rate 8000Hz/1000Hz
Init Gyro***

Accel Offsets X:0.00000000 	 Y:0.00000000 	 Z:0.00000000
Accel Scale X:1.000000 	 Y:1.000000 	 Z:1.000000
Gyro Offsets X:0.01297422 	 Y:-0.02389136 	 Z:0.00327944

Number of detected accels : 1
Number of detected gyros  : 1

Complete. Reading:

Menu:
    d) display offsets and scaling
    l) level (capture offsets from level)
    t) test
    r) reboot
r
3) After that, If I try to run ardupilot again, the Baro is detected and the software run without problem. If I reboot I need to do the same process to get ardupilot working,

**Issue details**
Ardupilot fail to initialise the INS

**Version**
Arducopter 4.0.5 and Arduplane 4.0.7 beta

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X quad and FX61 plane

**Hardware type**
Beaglebone green

**Logs**
root@beaglebone:~# config-pin -f agroraptor.cfg 
root@beaglebone:~# ./ardupilot/build/bbbmini/bin/arduplane -A udp:192.168.7.1:14550
MS5611 found on bus 2 address 0x01
MS4525: no sensor found
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver
Sensor failure: INS: unable to initialise driver

juvinski, 2020-10-10:
Hi,

after several tests, I discovery the faulty scenario.
The Baro and IMU is using SPI interface, for some reason, If I initialise the baro before the inertial sensor - both are using the same spi with spi_cs different (imu 0 and baro 1).
If I change the order - ins initialisation before the baro the things are working, but if I initialise the baro before the ins I got the problem.

any clue ?

Thanks

magicrub, 2020-10-11:
I'm not sure if this is related, but the baro will do a calibration on boot. This used to be a blocking operation but was changed to async about 2 years ago.

juvinski, 2020-12-22:
Hi,

the reason is something related to the way the device tree is done. With a DTB with all pins configured and kernel 4.14, everything works - just need to disable the boot overlays and cape manager.
With kernel 4.19 the PRU is not available yet.

closed 2020-12-22 by juvinski