ArduPilot/ardupilot #15555: Scripting: stuck in garbage collection when running out of memory

IamPete1, 2020-10-12:
A Scripting can get stuck in garbage collection. I have only been able to reproduce on real hardware (Cube Orange).  If running with SCR_DEBUG_LEVEL 2 you will see messages that the script is running until this happens, everything then just stops.  No 'no scripts to run' message and no out of memory warning. Increasing the SCR_HEAP_SIZE sufficiently resolves this.

https://github.com/ArduPilot/ardupilot/blob/e346cd3c8070f4397010d603c0b19be588674531/libraries/AP_Scripting/lua_scripts.cpp#L423

I was able to track down to that line using prints but have not looked further. This seems to require a reasonably complex script, a basic script just to just use all the memory does not work. I guess the trigger is actually the amount of memory that needs to be cleaned rather than the absolute memory usage.

This is the prints from my testing (its bottom to top because MP):
```
12/10/2020 2:06:05 PM : Lua: Ran
12/10/2020 2:06:05 PM : Lua: Running
12/10/2020 2:06:05 PM : Lua: Delay 19
12/10/2020 2:06:05 PM : Lua: start
12/10/2020 2:06:05 PM : Lua: GC done
12/10/2020 2:06:05 PM : Lua: Ran
12/10/2020 2:06:05 PM : Lua: Running
12/10/2020 2:06:05 PM : Lua: Delay 19
12/10/2020 2:06:05 PM : Lua: start
12/10/2020 2:06:05 PM : Lua: GC done
12/10/2020 2:06:05 PM : Lua: Ran
12/10/2020 2:06:05 PM : Lua: Running
12/10/2020 2:06:05 PM : Lua: Delay 20
12/10/2020 2:06:05 PM : Lua: start
12/10/2020 2:06:05 PM : Lua: GC done
12/10/2020 2:06:05 PM : Lua: Ran
12/10/2020 2:06:05 PM : Lua: Running
12/10/2020 2:06:05 PM : Lua: Delay 20
12/10/2020 2:06:05 PM : Lua: start
```

WickedShell, 2020-10-12:
`SCR_DEBUG_LVL` to 2 and what's the memory before trying GC?

IamPete1, 2020-10-12:
This is back on Master with default heap size (44,032), have to turn it up to ~100,000 before it will run correctly. (Bottom to top again)

```
12/10/2020 7:33:39 PM : Lua: Time: 41355 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 36746 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 38275 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 39212 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 46202 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 38253 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 39551 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 35902 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 35303 Mem: 35525 + 1010
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 36918 Mem: 35525 + 978
12/10/2020 7:33:39 PM : Lua: Running /APM/scripts/FenceMaker.lua
12/10/2020 7:33:39 PM : Lua: Time: 141882 Mem: 35509 + 1530

```