ArduPilot/ardupilot #15557: Add ability to skip waypoints during a mission without using a GCS

Lazydude, 2020-10-12:
If I exit Auto mode and "Cruise" to some point and then re-enter Auto mode I'd like to proceed to the nearest waypoint and continue from there.  Don't continue from the last active waypoint but continue from the next nearest waypoint - skipping waypoints I've cruised past. 

Previous discussion in the forum, link to it:  https://www.rcgroups.com/forums/showpost.php?p=45633741&postcount=5211

**Describe the solution you'd like**
I believe this would require 2 new options;  one to enable it and one to designate the distance required to consider a waypoint executed or no longer desired.

For example with the MIS_RESTART parameter it has either Resume or Restart.  Couldn't an option be added such as "Resume Auto Missions from Nearest Waypoint" so I don't have to travel back to unexecuted waypoints that may be over a mile back in the auto mission and do it without a GCS.

If I've flown xx meters or feet past an un-executed waypoint then skip it and continue from the next nearest waypoint. Don't fly backwards in the mission - only forward and do this without a GCS.

**Describe alternatives you've considered**
There doesn't seem to be any alternatives outside of using a GCS to do this.  If I flew from my house this wouldn't be an issue but I'd rather not lug all that equipment out to a remote flying location.  Many others voice agreement that this would be nice option to have.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

Thanks for considering this.

rmackay9, 2020-10-12:
resuming from the the closest waypoint or closest line between waypoints would be a nice feature.

lazarovtsvetozar, 2020-10-13:
I all agree with this feature. It will be nice to have this option!