ArduPilot/ardupilot #15558: Disable autonomous descent when using Terrain Following

Lazydude, 2020-10-12:
Take the following example:  I'm flying in cruise mode with terrain following (TF) enabled.  If I start flying along a big river channel with lots of side channels I get terrible bobbing as I go along.  It would be nice to have an option with TF that it doesn't descend but only climbs if necessary to avoid obstacles. 

I believe this could be done with only one added option - "Terrain Descend"  Disable autonomous descent.  Stay at current height unless climbing is necessary to avoid an obstacle.

**Describe alternatives you've considered**
The only alternative is to disable TF altogether.  I consider this a better alternative than the "Non-Auto Terrain Following" option 86 that is still being considered.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

See picture for example of previous flight exhibiting this.
[image]

Naterater, 2020-10-19:
While this could be added as an enhancement, I think that for many users including myself, the descending is necessary to meet regulations of altitude ceilings that reference the ground. It is also used to keep a camera at constant altitude above the ground.

magicrub, 2020-10-19:
@Lazydude have you tried changing param [TERRAIN_LOOKAHD](https://ardupilot.org/plane/docs/parameters.html#terrain-lookahd-terrain-lookahead)?

iNotMario, 2024-08-01:
Time for a bump - because i got a bit of a scare flying over a hilly valley. :D
I second this request ... or endeavor for a better option, as the sentiment is valid.

This could be particularly useful for Failsafe RTL scenarios - where the last thing you want is for the plane to descend - worsening the comm situation.

Perhaps a TERRAIN_AVOID bitmask would be a good way to represent this - acts the same as TERRAIN_FOLLOW for the modes selected, except only climbs to maintain AGL, never descends.

I'm still looking at the code to figure this one out, but if i understand the way it works correctly, it would effectively be a hybrid of Absolute Reference and Terrain Reference: Altitude error would be computed both ways, and the minimum value returned. Still looking into how i can get it to compute both errors simultaneously, as it looks like switching the alt_frame destroys the prior value of the prior alt_frame.