ArduPilot/ardupilot #15562: Copter: Internal errors (0x4000)

Notorious811, 2020-10-13:
## Bug report

**Issue details**
Was testing HERE+ RTK GPS setup on 5" quadcopter running arducopter v4.0.4 on omnibusf4pro, so i left the drone outside for more than 1 hour to see the status of RTK GPS. Green LED was blinking in the starting and i didn't expected any change of color of LED status, i was not keeping an eye throughout that period, so after around 1 hours and 30 minutes, when i come back to see the status then i saw yellow LED blinking, i connected the drone to mission planner and found this error **PreArm: Internal errors (0x4000)**  on HUD.

This the 4th different kind of internal error which i have seen now in last 1 month, i had log_diarm enabled, but i doubt if anything can be confirmed from the log and similar to its other internal errors brother, this one also not reproducible and went away after reboot.

**Version**
ArduCopter v4.0.4

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad-X

**Hardware type**
Omnibusf4pro

**Logs**
https://drive.google.com/file/d/1GHETwCll1ou7_h-SX5gj8PsMFr4NrLtk/view?usp=sharing

[image]

leonisflying, 2020-11-03:
I ran into same problem. I found it inevitably comes out when I upload over 305 waypoints.
My guess, it's a kind of memory overflow.
My fc is a Matek f405wing, planning to order a matek f765wing which has bigger RAM & flash, 
so I can find out if it's a memory issue.

muramura, 2020-11-07:
If it's due to a lack of free memory space, then I think it would be better to implement a feature that monitors the free memory space and triggers a failsafe if it continues to decrease.

Notorious811, 2020-11-07:
in the context of free memory space, i have another issue, i have disabled smartRTL and terrain features on omnibusf4pro, after disabling these features i got around 55Kb of free memory from total 192Kb memory on omnibusf4pro board, so i thought to do some experiments with that free memory as i thought memory is a kind of resource which can be good to have if you wanna try some firmware customizations, but in case where i use this free memory space to create string buffers in heap or in static memory space then there seems some issues if i use more than 16-17Kb of free 55Kb memory, using more than 17Kb of 55Kb memory space for string buffers interrupting Neopixel LEDS on servo7/servo8 on omnibusf4pro and they stop working, so there's still more than 35Kbs of free space available and i cannot allocate more string buffers as it causing problems for Neopixel LEDs

yatink-ci, 2023-04-12:
[image]

gogowitsch, 2025-06-08:
This error briefly comes up in this Spiderweb footage: https://youtu.be/8JjC_119WUA?si=FKwUW9KURx-zHd7R&t=190