ArduPilot/ardupilot #15563: VOLZ/Sbus Protocol sending over more than one UART port at the same time!?

lazarovtsvetozar, 2020-10-13:
[https://discuss.ardupilot.org/t/volz-protocol-over-2-different-uart-ports/62681]
-- Feature request
VOLZ/Sbus Protocol sending over more than one UART port at the same time.

-- I have found an issue with creating a redundant RS485 Servo Line with VOLZ Protocol. When I set up more than one UART port for VOLZ protocol use the autopilot sends data over only one of the UART ports while the other is inactive.  I have also noticed that the UART port which gets the signal is the one having the lower numerical value preset in Arduplane. -- 

-- I would like to have the option to use more than one UART port for VOLZ protocol or SBus protocol in order to make redundant physical RS485 Servo Line.  -- 

--Platform--
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

[image]

peterbarker, 2020-12-08:
It would be relatively straight-forward to send the same data to all channels marked as volz - iterate through all marked, push into an array, `port->write(...)` the data to each.

However, this might cause issues down the line if we wanted to send higher-numbered servos later.

Might take this to a devcall.

tridge, 2020-12-14:
@peterbarker I agree, sounds good

peterbarker, 2020-12-16:
@lazarovtsvetozar are you planning on attacking this problem?

lazarovtsvetozar, 2020-12-17:
Hi Peter (@peterbarker),
Unfortunately I would not be able to handle this, because my knowledge is  really low in programing and I do not think I could do it However, if I can help you out in other way I`l do it
Have a nice day !