ArduPilot/ardupilot #15573: RFC: Autotest splitting

khancyr, 2020-10-14:
## Feature request

**Is your feature request related to a problem? Please describe.**
The sum of our test is long and make CI long to validate a PR. We could improve this by automatically splitting the tests into timed group of test and profit from our parallelism in CI.

**Describe the solution you'd like**
Currently some autotests are taking long time, like for copter more than 40min. We could improve our architecture in some way.

We could manually made some timed group of test but this will need some maintenance as we continuously push new tests.
Therefor I propose to implement a way to split the test automatically.

Here is what I think of : 

On each test, we put an estimate completion time, rounded at the upper min. This can be easily obtain by launching the tests individually. Our test are mostly deterministic, that mean that for the same input we get the same output, so we shouldn't have huge timing variation across run.

We implement a `--split X` option on autotest.py. This option will parse the timing of the tests and use a solver to make timed groups of test that shouldn't be longer, in total, of X min. This should write the test group into a json file. Why use a file instead of direct command ? I doubt that we could have deterministic solver to make the tests groups, so that is better to generate the group once and use the file to have the group definition. 

After the `split` option, we should have the `--group X` option that will as input the split json, and execute the tests from the group X

This combination of option should allow us low maintenance against the manual group method and will work the same accross all vehicle.

Optional : add a time watchdog for test. If a test take 30% more time that the expected time then we got an issue and fail the test.

Comments welcome

davidbuzz, 2020-10-14:
I propose we do this simple and immediate fix to bring github actions CI autotests down to around , currently,  13minutes... https://github.com/ArduPilot/ardupilot/pull/15570  just merge it, its tested and works now. Following on from this , you can implement a smarter solution without needing to rush,etc.

peterbarker, 2020-10-14:
One issue I do have with moving tests around - the unknown dependencies between tests that cause either flat-out failures (easy to work with), or intermittent test failures (nasty).

An example of that is here, in the split-tests-to-make-things-go-really-fast PR: https://api.travis-ci.com/v3/job/399293391/log.txt

We already have a significant problem with intermittent test failures, and they lead to people getting frustrated with CI when tests fail that are completely unrelated to what they're trying to get done.

History has shown quite clearly that there isn't a general appetite for going and fixing these intermittently failing tests.  People gets bothered by them, people think it is someone else's job to go and fix them.

Tests should pass regardless of what comes before them.  They don't.  That should be fixed, of course, but as I say - people want to do the stuff they're working on, not hare off on a mostly unrewarding bug-hunt.

davidbuzz, 2020-10-16:
Personally, the main reason/s i avoid fixing "autotests" in general is that they:
- for-the-most-part work well
- for-the-most-part critical issues get fixed by @peterbarker faster than I can work.     
- "giant wall of text" that is produced as the log/s for each CI run are too big and too scary to go deep diving into
- when autotest/s are bundled together into code that takes 40+mins to run, its hard to know where to start.
 -unless u 'deep dive' its super hard to even find out which autotest/s failed - it would be nice to get an overview that's easier to consume.
My opinion is that, at least for github actions, we are now in a position where we can tweak things to make it more accessable, which includes making the test chunk/s smaller , and I'm hoping that we could maybe make the output log/s NOT generate huge output on the part/s that are successful.

peterbarker, 2024-07-31:
@davidbuzz actually did this for us a long time ago.  It has worked well over the years, thanks Buzz!

Closing this one now.

closed 2024-07-31 by peterbarker