ArduPilot/ardupilot #15574: Airsim Simulation - [Errno 111] Connection refused sleeping

bilginmu, 2020-10-14:
I try to use Airsim with ardupilot. I downloaded a precompiled Airsim binary(Africa). First I ran Airsim with below command
 ./Africa_001.sh -ResX=640 -ResY=480 -windowed
and then I started ardupilot with 
sim_vehicle.py -v ArduCopter -f airsim-copter --console --map
I get the error below:
[image]

closed 2021-07-26 by IamPete1

Abner0907, 2021-08-12:
Do you know what 's the problem? I have the same issues.

hendjoshsr71, 2021-08-15:
@Abner0907  See here: https://ardupilot.org/dev/docs/building-setup-windows10.html#using-sitl-with-wsl-and-xwindows

"Install an XWindows application to run graphical programs such as SITL by installing VcXsrv, Cygwin X or Xming on Windows."

If we can make something clearer here let me know.