ArduPilot/ardupilot #15579: Autotest fails to detect reboot if FORMAT_VERSION is present in param file

sypaq-nexton, 2020-10-15:
## Bug report

**Issue details**

If SITL is loaded with default params that include the `FORMAT_VERSION` param, reboots of SITL will erase the EEPROM. The consequence of this is that Autotest won't detect the reboot and will fail. 

To reproduce, the sequence of events is:
1. Autotest loads the ArduPilot SITL executable with both the `-w` flag and default param values (which include the `FORMAT_VERSION` param).
2. The SITL EEPROM is erased
3. SITL writes the `FORMAT_VERSION` param to EEPROM, but because this value is the same as the default value loaded in, the _write does not occur_.
4. The Autotest commands a reboot (recording the current value of the `STAT_BOOTCNT` param)
5. SITL reboots.
6. SITL attempts to read the `FORMAT_VERSION` param from EEPROM, but it's not present (see step 3).
7. As the `FORMAT_VERSION` param cannot be verified, SITL erases the EEPROM. This resets the `STAT_BOOTCNT` param.
8. Autotest checks the value of the `STAT_BOOTCNT` param, but as it is not the expected value, it throws an exception to fail the test.

This is very edge-case behaviour, but it bit us, so I'll capture it here.

**Version**
Plane 4.0.3

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
SITL, Autotest

**Logs**
_Please provide a link to any relevant logs that show the issue_

sypaq-nexton, 2020-10-15:
The obvious work-around is to remove the `FORMAT_VERSION` param from the param file. This bit us because we pulled full param files from a vehicle (which include the `FORMAT_VERSION`) then used them for SITL testing.

sypaq-nexton, 2020-10-15:
A potential code fix is to replace the following line in `Parameters.cpp`:
```cpp
g.format_version.set_and_save(Parameters::k_format_version);
```
with:
```cpp
g.format_version.set(Parameters::k_format_version);
g.format_version.save_sync(true);
```
to force the `FORMAT_VERSION` param to be written.

peterbarker, 2020-10-15:
On Wed, 14 Oct 2020, Nick Exton wrote:

> The obvious work-around is to remove the FORMAT_VERSION param from the param file. This bit us because we pulled full param files from a vehicle (which
> include the FORMAT_VERSION) then used them for SITL testing.

Should probably just throw an exception in autotest in the case that it is 
present in a parameter file.

Resetting FORMAT_VERSION is a relatively common way for users to reset 
their parameters to default - even 'though we now have a mavlink message 
to do so.

sypaq-nexton, 2020-10-15:
> Should probably just throw an exception in autotest in the case that it is present in a parameter file. Resetting FORMAT_VERSION is a relatively common way for users to reset their parameters to default - even 'though we now have a mavlink message to do so.

That will partially resolve this issue, but won't fix it if you call SITL directly. Sorry, I wrote it up as an autotest problem because that's the easiest way to reproduce.

The core issue is with SITL. Starting SITL with the combination of:
1. the `-w`/`--wipe-eeprom` flag set; and
2. a `--defaults` param file with the `FORMAT_VERSION` defined as the `k_format_version` value (e.g. `13` for Plane);

will result in an invalid EEPROM file that will cause SITL to erase EEPROM on the next boot:
1. the `-w` flag causes the SITL wrapper to erase the Params (via a call to `AP_Param::erase_all()`)
2. the vehicle is initialised, and a call to `load_parameters()` is made
3. the `FORMAT_VERSION` param can't be found, so the entire EEPROM is erased
4. the `FORMAT_VERSION` param is set to `k_format_version` and saved, but because it's the same as the default, nothing is written.

Subsequent executions of SITL will not find the`FORMAT_VERSION` param in the EEPROM file, assume it is invalid, and erase the EEPROM again.

Like I said, this is edge-case behaviour, but it's easy to hit if you run SITL with param files pulled directly from a vehicle.

peterbarker, 2020-10-15:
On Wed, 14 Oct 2020, Nick Exton wrote:

> Like I said, this is edge-case behaviour, but it's easy to hit if you run SITL with param files pulled directly from a vehicle.

Yep.  I'm really loathe to adjust cross-platform code because of the edge 
case in SITL.

We could explode where we parse the default parameter file within SITL if 
we come across FORMAT_VERSION - just panic or exit 1 or something, nothing 
neat - just make it clear to the user that it's invalid to include that in 
a parameter file.

sypaq-nexton, 2020-10-15:
> Yep. I'm really loathe to adjust cross-platform code because of the edge case in SITL. We could explode where we parse the default parameter file within SITL if we come across FORMAT_VERSION - just [...]

Perhaps just ignore the `FORMAT_VERSION` in SITL, rather than panic? E.g., could drop the following into `AP_Param::parse_param_line()`:
```cpp
#if CONFIG_HAL_BOARD == HAL_BOARD_SITL
    // Workaround to prevent FORMAT_VERSION in param file resulting in invalid
    // EEPROM. For details, see: https://github.com/ArduPilot/ardupilot/issues/15579
    if (strncmp(pname, "FORMAT_VERSION", 14) == 0) {
        ::printf("Warning: Ignoring FORMAT_VERSION in param file\n");
        return false;
    }
#endif
```

peterbarker, 2020-10-15:
On Wed, 14 Oct 2020, Nick Exton wrote:

>       Yep. I'm really loathe to adjust cross-platform code because of the edge case in SITL. We could explode where we parse the default parameter
>       file within SITL if we come across FORMAT_VERSION - just panic or exit 1 or something, nothing neat - just make it clear to the user that it's
>       invalid to include that in a parameter file.
> 
> Perhaps just ignore the FORMAT_VERSION in SITL, rather than panic? E.g., could drop the following into AP_Param::parse_param_line():
> 
> #if CONFIG_HAL_BOARD == HAL_BOARD_SITL
>     // Workaround to prevent FORMAT_VERSION in param file resulting in invalid
>     // EEPROM. For details, see: https://github.com/ArduPilot/ardupilot/issues/15579
>     if (strncmp(pname, "FORMAT_VERSION", 14) == 0) {
>         ::printf("Warning: Ignoring FORMAT_VERSION in param file\n");
>         return false;
>     }
> #endif

Definitely worth a PR!

We'll chew it over at the next devcall, but I'd be happy enough with that.

sypaq-nexton, 2020-10-20:
Addressed in PR #15583.

closed 2020-10-20 by sypaq-nexton