ArduPilot/ardupilot #15588: Copter: Watchdog in-flight resets

jperez-murcia, 2020-10-16:
**Issue details**
Watchdog reset controler during flight condition; disabling watchdog and tested without arming motors, ends in a hanged controller.

When at enough altitude, two logs appear : first one ended at top flight height, and second one starts five second later, and several meters down, until 0m crash

**Version**
Arducopter 4.0.4 & 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Radiolink Minipix v1.2

**Logs**
[first flight, first part](https://drive.google.com/file/d/1JAJcGeehRUfTDplApMaY6p8K62FVy4Ox/view?usp=sharing)
[first flight, last part](https://drive.google.com/file/d/106LuNIx3u1vtdgpEYw3ojk7bke7KhsNt/view?usp=sharing)

[second flight, first part](https://drive.google.com/file/d/19CdwHZm_y8LI1sVdY9VWRjOJ4NM92UkQ/view?usp=sharing)
[second flight, last part](https://drive.google.com/file/d/1ioGBacwu6EBmuz80lCPPPk1nLDTlAU0k/view?usp=sharing)

amilcarlucas, 2020-10-17:
This bug is probably fixed in 4.0.5-rc2. Can you please test if it fixes this issue for you?

auturgy, 2020-10-20:
WDOG.FT (fault type) = 3, which is a hardfault.  Noting that the watchdog was added specifically due to the MiniPix, I wouldn't be at all surprised if this is a legitimate board hardware error.

jperez-murcia, 2020-10-20:
> 
> 
> This bug is probably fixed in 4.0.5-rc2. Can you please test if it fixes this issue for you?

I'm on the way .... DEVEL 4.0.5 firmware flashed to several boards; watchdog disabled to be sure about hangs .... up to now, 10h continous running (not flying) with no hungs ... let's wait 20 or 30 more hours ....

jperez-murcia, 2020-10-20:
> WDOG.FT (fault type) = 3

auturgy, please, where did you find this info ?

IamPete1, 2021-07-26:
Sounds like this was resolved.

closed 2021-07-26 by IamPete1