ArduPilot/ardupilot #15589: Make SBUS able to output the two digital SBUS channels

blacknose2010, 2020-10-16:
**Is your feature request related to a problem? Please describe.**
For my application I need digital SBUS outputs as in bit 7 and 6 of the SBUS _flag_ byte. (second last byte of the SBUS frame)

**Describe the solution you'd like**

Add parameters like _Servo_SBUS_digital_out_1_ and _Servo_SBUS_digital_out_2_ including each Servo functions, min and max PWM threshholds for high and low etc. 

**Describe alternatives you've considered**
no idea for alternative

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._
SBUS frame description e.g. here https://github.com/bolderflight/SBUS#description

IamPete1, 2020-10-17:
What is your use case? A possible solution would be to assign then to relays. Or give scripting the ability to set them. Using servo functions for a boolean output is a bit strange.

blacknose2010, 2020-12-28:
The use case was actually a bad workaround idea to control a different closed source autopilot that uses SBUS IN for RC (daisychanied) and would require those bits 7 and 6 in the SBUS flag byte. However, this idea has luckily been discarded by now and this ticket can be closed.

closed 2020-12-28 by blacknose2010

magicrub, 2020-12-29:
Which closed source autopilot?