ArduPilot/ardupilot #15595: UI obscured not easy to use on Parallels / Windows

reduxdj, 2020-10-18:
## Bug report

**Issue details**
Parallels, UI Overlaps and is hard to use
_Please describe the problem_

**Version**
1.37563

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

[image]

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

amilcarlucas, 2020-10-18:
This is a mission planner issue. Not an Arducopter firmware issue. Please report it to mission planner github.

closed 2020-10-18 by amilcarlucas

magicrub, 2020-10-19:
@reduxdj here's the link https://github.com/ArduPilot/MissionPlanner/issues/new