ArduPilot/ardupilot #15608: Plate: Automatic landing with an enabled rangefinder causes immediate full descent into the ground

Naterater, 2020-10-20:
## Bug report
**Issue details**

Plane lands excellently using RTK. As soon as we plug in a rangefinder, the landing approach is hijacked and the airplane descends at full rate to the ground. RNGFND_LAND = 0 the whole time. Very repeatable. 

**Version**
4.0.7 beta 2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin motor electric with reverse thrust

**Hardware type**
Cube orange, BlHeli ESC's with telemetry, mRo F9P GNSS with NTRIP RTK streamed

**Logs**
RTK successful landing AND Rangefinder plugged in landing:
https://1drv.ms/u/s!AunBK4rE1ZDCg4g8p77yzpds4a_oRg?e=XC341p

peterbarker, 2020-10-20:
@Naterater 

[image]

Any idea what's going on there?  Those seem to be abnormally low readings for chunk of level flight?

Naterater, 2020-10-20:
Don't get me wrong, we have issues to work out with the rangefinder providing better data, but until we enable that rangefinder for anything in the parameters... for example RNGFND_LANDING, I don't understand why any of that should matter.

tridge, 2020-11-02:
@Naterater thanks for reporting this, I finally found the issue and I believe it is fixed here:
https://github.com/ArduPilot/ardupilot/pull/15710
there were two places where rangefinder data could affect landing when RNGFND_LANDING=0

PR #15710 (2020-11-02): Plane: fixed usage of rangefinder in landing with RNGFND_LANDING=0
this fixes two places where rangefinder can affect landing when
RNGFND_LANDING=0. If rangefinder data was invalid then this could cause a bad landing slope

This fixes #15608 

Many thanks to @Naterater for reporting!
closed 2020-11-02 by tridge