ArduPilot/ardupilot #15617: asserts in non-debug builds

sypaq-nexton, 2020-10-20:
## Bug report

**Issue details**

There are a number of `assert`s in the codebase. Normally, in a production build, these `assert`s would be compiled out using the `NDEBUG` preprocessor macro. However, in ArduPilot, `assert`s are enabled in both production and debug builds. 

Is this by design, or is this something we should resolve?

**Version**
All versions.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

sypaq-nexton, 2020-10-20:
@tridge: FYI

magicrub, 2020-10-20:
@sypaq-nexton in many cases the asserts and other checks are only compiled with SITL so that developers and CI and catch it at runtime but not be in flown firmware.
Example:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass.cpp#L661-L663

Can you link us to some asserts that you have in mind?

sypaq-nexton, 2020-10-21:
@magicrub: The ones I was thinking of specifically are `assert()`, not `AP_HAL::panic()`:
* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor_Invensense.cpp#L936
* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_AK8963.cpp#L331

There are a few more scattered throughout the code, but they're mostly either in test code or in initialisation code.

WickedShell, 2021-04-13:
Asserts were removed between #17002 and #17004

closed 2021-04-13 by WickedShell