ArduPilot/ardupilot #15622: SITL: synthetic clock is used even without the --synthetic-clock option

ntamas, 2020-10-21:
## Bug report

**Issue details**

It seems like the `--synthetic-clock` option of the SITL simulator has no effect -- the value of `_synthetic_clock_mode` in `SITL_cmdline.cpp` is set to `true` unconditionally when the `sitl_model` is constructed, which basically means that the SITL simulator always uses a synthetic clock.

This is not necessarily a bad thing, but the presence of such an option can be confusing to new users -- I've spent a bit of time trying to figure out what this option does. I would be happy to send a PR that removes this command line option entirely, or at least removes it from the list of documented options and prints a warning when someone still uses it (in order not to break any scripts that might rely on the presence of this option).

**Version**

4.0.3, but it's still present in the head revision of the `master` branch at the time of writing.

**Platform**

[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Not applicable.

**Hardware type**

SITL simulator.

anirbansaha782, 2020-10-23:
I am new to this contribution. Can anyone plz help me where to look up of this issue.?? I am unable to locate the SITL_cmdline.cpp file.
Thanks in advance

ntamas, 2020-10-24:
It is in `libraries/AP_HAL_SITL`.