ArduPilot/ardupilot #15638: Copter/Rover: "external" failsafe triggered by a "button"

rmackay9, 2020-10-24:
It would be convenient for some users to be able to trigger a failsafe action (i.e. RTL, SRTL or Land) by providing a voltage into the autopilot (i.e. 3v = OK, 0 = bad).  I think this could be accomplished by doing the following:

- add a new FS_EXT_ENABLE or FS_EXT_ACTION parameter with some value, 0=Disable, 1=RTL, 2=Land, etc (try to use the same enumerations as we use for other failsafes)
- add a new FS_EXT_PIN parameter would allow the input pin to be defined.  It's possible it could always use a Button in which case the options might be Button1, Button2.  Doing it this way would lead to some extra setup for the user though as they would then need to setup the BTN_ parameters.  It might save some code however because AP_Button probably has some debounce logic built-in.
- another parameter might be required to allow the user to define whether a high voltage or low voltage triggers the failsafe although in general, a high-voltage for OK is safer.

If the failsafe should probably both block arming and trigger a failsafe.

Another possible solution is that this should all be done using Lua scripts which have the ability to read from buttons and block arming.

auturgy, 2020-10-24:
I’d like to see this use ap_button.
The use of scripts for Failsafe requires either assurance that the script is always running - which is not true by design - or an additional “script failsafe” to be added, such that a failsafe action is triggered by script failure.

https://ardupilot.org/copter/docs/common-lua-scripts.html#script-crashes-and-errors

IamPete1, 2020-10-24:
Will be quite easy with https://github.com/ArduPilot/ardupilot/pull/15484

I think external fail safe is quite a nice idea, could be triggered from button or RC or scripting.

geofrancis, 2021-09-10:
Rather than this being strictly a failsafe button, why not just treat it like an RC channel and use it to trigger the list of Auxiliary Functions? something like ext_btn = 4 to have it enable RTL or 31 for motor emergancy stop, it could be used by external controllers to trigger lots of things without complicated code.

peterbarker, 2021-09-10:
@geofrancis we allow auxillary functions to be triggered via AP_Button now.

We don't use any failsafe mechanism (to trigger disarms and alerts or whatever), but we have a failsafe triggered by an AuxFunction being asserted and failsafe cleared when it is de-asserted.

Really need @rmackay9 to specify any additional functionality required to satisfy this requirements here.

rmackay9, 2021-09-10:
I think I wrote this issue at the request of AION robotics.  Lua scripts are so capable now though I'm not sure if we really need an external "failsafe" anymore.  What is a failsafe except a specific action like RTL, Land, etc..?

geofrancis, 2021-09-10:
@peterbarker good to know, I have been out the loop for a couple of years due to medical reasons. im blown away with everything that has been added and its been a lot of reading trying to catch up, last time i looked all you could do was enable/disable the motors with a button i didn't realize so much had been done already.

peterbarker, 2021-09-13:
Sounds like we can close this one as "Scripting" :-)

closed 2021-09-13 by peterbarker