ArduPilot/ardupilot #15648: magnetic interference due to IMU heater

tridge, 2020-10-26:
The magnetic interference due to the IMU heater on CubeOrange (and likely other boards) is very large.
[image]
On Z axis the heater induces a 200 mGauss change. On X/Y it is about 20 mGauss.
The Z axis change would be enough to cause EKF to stop fusion as it is beyond the typical mag innovation gate.
We should add compensation for this in AP_Compass, using fixed gains in hwdef.dat and a hal.util accessor for the heater duty cycle.
Most users don't use the AK09916 on the CubeOrange, but some do, and at the moment if they flew into a cold patch of sky their compass could start being rejected by the EKF.

zyromskij, 2020-11-08:
Couldn't we just redesign the heater? I'm assuming that all of this comes from the power switching noise? If yes then why not use a controlled, linear current source to heat up the resistors?

JoshWelsh, 2020-11-08:
> Couldn't we just redesign the heater? 
I don't think that would do anything for the number of Cubes out there today.

zyromskij, 2020-11-08:
I know it's impossible to implement it in those that are out there but it could be done in new Cube versions I guess.

charleslinquist, 2020-11-08:
I don't know the layout of the CUBE, but does anyone know if the magnetic field is caused by the wires that the feed the heater, or is it the heater itself?  If it is the wires that feed the heater, they could be rerouted.

mice5, 2020-11-17:
> I don't know the layout of the CUBE, but does anyone know if the magnetic field is caused by the wires that the feed the heater, or is it the heater itself? If it is the wires that feed the heater, they could be rerouted.

Heater is just 5 x 5-Ohm resistors + BJT transistor routed in "coil" around the IMU board. That's why it mainly affects the Z axis.

> Couldn't we just redesign the heater? I'm assuming that all of this comes from the power switching noise? If yes then why not use a controlled, linear current source to heat up the resistors?

It would only make the interference continuous, but still changing depending to temperature and driven current.

To make this design work, i would route the heating differentially in pairs in both directions, so magnetic fields cancels out by itself. Or make some kind of resistive uniform layer (like those heating 3d printer pcbs) to make the heater more similar to heating plate.

Another approach would be turning off the heater and use temperature compensation algorithm for sensors, if the user wish to use built-in compass.

Btw. what are the benefits of the heater, if it takes long minutes for IMU to heat up to desired temperature and users take-off earlier. Then, what if all the calibrations were done in different conditions? IMO, it creates as many problems as it solves.

mcsauder, 2021-02-16:
>To make this design work, i would route the heating differentially in pairs in both directions, so magnetic fields cancels out by itself. Or make some kind of resistive uniform layer (like those heating 3d printer pcbs) to make the heater more similar to heating plate.

That's the right approach in my experience.  The EM field generated by the power traces can be stronger than the earth magnetic field being measured, (~0.5G), when the traces are close (<= ~1cm), so the careful design of the resistive elements and balanced and closely coupled power traces is critical (and requires hardware validation of the design layout).  Another point to watch is the radiative energy from a resistor striking only one side of an IMU or mag die.  This can create issues of localized stress/strain in the silicon, so placing a thermal pad or something similar to block radiative energy also needs to be considered.

>Btw. what are the benefits of the heater, if it takes long minutes for IMU to heat up to desired temperature and users take-off earlier. Then, what if all the calibrations were done in different conditions? IMO, it creates as many problems as it solves.

Systems I've been contributed to typically see a ramp from ambient to ~55C in 7-10 seconds, usually before the system is fully initialized and before GPS lock is usually obtained.

The plots shown initially are clearly driven by the EM field being generated by the resistive heating circuit, so that's going to be difficult to get around as it is intrinsic to the PCB layout/design.

AlexKlimaj, 2021-02-16:
@mcsauder and I have a lot of experience with heaters and mags. We are current working on two different fmuv5 hardware designs with heaters. 
Here is a log where the mag is about 4cm from the heater and roughly in the same plane. I start the heater driver at 1:20. You can see the mag jump when the heater is on 100%. When it stabilizes, you can see the mag data is more noisy. 

https://review.px4.io/plot_app?log=e0744744-0559-473f-be5c-7874505fd490

[image]

This is a different vehicle with two mags. The "external" mag is about 4.5cm away and above on a different board by about 1cm. There really is no impact on the mag data in this case.
https://review.px4.io/plot_app?log=2d77e25b-b8d5-4b41-bb66-a5ad94f25754

IamPete1, 2021-07-18:
closed by https://github.com/ArduPilot/ardupilot/pull/16551

closed 2021-07-18 by IamPete1