ArduPilot/ardupilot #15651: EKF3: emergency yaw reset is triggered if GPS is disabled

rmackay9, 2020-10-26:
The GSF emergency yaw reset seems to be triggering a yaw reset if the GPS is disabled.

This can be re-created in SITL by doing the following:

- start Copter in SITL
- enable EKF3
- arm and takeoff in Loiter
- param set GPS_TYPE = 0

After a few seconds, "EKF3 IMU0 emergency yaw reset" will be displayed on the screen.  This seems incorrect because there is no GPS data available for the GSF feature to work from.  It also leads to a position and velocity reset.  The velocity reset leads to the EKF's velocity estimate jumping to zero as shown below.

[image]

priseborough, 2020-10-28:
What happens if GPS is stopped externally to the EKF, ie GPS_TYPE = 0. The EK3_GPS_TYPE parameter wasn't originally developed to be changed during operation.

priseborough, 2020-10-28:
I'm now also having trouble replicating the result using EKF_TYPE = 0 to stop GPS data, but have seen a place in code where  race condition could result in the EKFGSF yaw estimator validity remaining high for a short period after GPS data stops.

priseborough, 2020-10-28:
The result can be replicated by using RC inputs to build up horizontal velocity before turning off GPS.

priseborough, 2020-10-28:
So what is happening is that the Copter::ekf_position_ok() function is returning false when the EKF has been without GPS  for too long which causes Copter::ekf_over_threshold() to go true which causes ahrs.request_yaw_reset() to be called. The EKF then honours the reset request.

We shouldn't be asking for a yaw reset if innovation levels are OK and Copter::ekf_position_ok() is false becasue that indicates the loss of nav is due to no GPS, not bad GPS fusion.

rmackay9, 2020-10-30:
resolved in master with PR https://github.com/ArduPilot/ardupilot/pull/15670

closed 2020-10-30 by rmackay9