ArduPilot/ardupilot #15658: Air Density and pressure being affected by barometer temperature

fr-correia, 2020-10-27:
## Bug report

**Issue details**
Hi guys, I marked this as a bug, but I'm not sure if should have marked it as a new feature.
So here is the deal, I have seen situations in which the aircraft comes very very close to a stall, and never really got an answer for that. Recently I have been looking at barometers, because I noticed that sometimes they are affected by temperature, especially if my board is in a place where it heats a lot, which is easily spotted when the altitude from the barometer just starts changing a lot for a fixed position of the aircraft.
So I digged in a bit and noticed that the Arducopter project has a Barometer temperature calibration. Should we expect the same algorithm on the Arduplane any time soon?
Also, there is a catch... If the temperature on the board might affect altitude values, it should also affect the air density, am I right? And if it does, it might affect the scale factor that converts from equivalent to true airspeed!

**Just a quick side note:**
The barometers altitude is always calculated using the ground temperature. Is there a specific reason on why not just use the standard lapse rate from the ISA, since it is used globally in all aircrafts?
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Pixhawk

magicrub, 2020-10-27:
@fr-correia The baro on copter is the same as plane. Where do you see it acting different?

fr-correia, 2020-10-27:
@magicrub Yes baro is the same for copter and plane, except for the use of the AP_TempCalibration library. This library is not included in the Arduplane. 
The documentation on how to use it in copter is presented here:
https://ardupilot.org/copter/docs/common-baro-temp-comp.html

WickedShell, 2020-10-29:
We calculate with the standard lapse rate being assumed from the ground reference point. Barometer temperatures are a problem, primarily because we are subject to a lot of self heating. If your airspeed sensor provides a temperature we will actually use this instead as it's presumed to not be mounted so close to all the other flight controller bits that are producing as much heat. You can also directly specify the reference temperature.

closed 2020-10-29 by WickedShell