ArduPilot/ardupilot #15660: Change RNGFND parameters to "enable" style

xfacta, 2020-10-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
Many RNGFNDx parameters show up in configurable parameters and in logs when a rangefinder is not even in use nor configured, leading to unnecessary parameter clutter and extra logging.

**Describe the solution you'd like**
Use RNGFNDx_TYPE as an "enable" setting to allow the rest of the rangefinder parameters to be visible and usable. A similar example would be the BATT parameters.
By default RNGFNDx_TYPE is already set to 0. This could prevent the entire suite of RNGFNDx parameters appearing in the groundstation and in logs.

**Describe alternatives you've considered**
Alternative is to leave it as-is.

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2020-10-28:
I have a PR that fixes this https://github.com/ArduPilot/ardupilot/pull/15227

xfacta, 2020-10-29:
Thanks @amilcarlucas - do we just close this enhancement/feature request? or do we wait till your changes are merged and tested?

amilcarlucas, 2020-10-29:
Wiat until it is merged to close.

IamPete1, 2021-07-26:
This was fixed by https://github.com/ArduPilot/ardupilot/pull/14560

closed 2021-07-26 by IamPete1