ArduPilot/ardupilot #15666: [VTOL] QRTL: quite castrating stopping point caculation

robustini, 2020-10-28:
I'm doing some tests on Realflight with the tailsitter, the RTL_RADIUS in case of Q_RTL is absolutely not respected.
 I'm varying from 1 to 20 meteres, the transition FW-> MC always occurs at the same distance from the home, so the transition distance from home is too high and the tailsitter remains in copter a lifetime, not good.
In my opinion this is fine with standard quadplanes, where maybe the user drops the radius too much and exceeds the home, but with the tailsitter no, especially not vectored, at least this I'm experiencing.
I would prefer to have a precise intervention on that parameter, so that I can center it according to the configuration of the VTOL, just do transitions tests and check the distance margin.
I therefore believe that the stopping point caculation must be removed, or that we can intervene on the RTL_RADIUS at a low level in case of tailsitter, or of VTOLs that perform the transition in a few meters.

**Version**
Tested with the latest master.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tailsitter non-vectored

robustini, 2020-10-28:
I tried everything @IamPete1 and I never got a decent result.
The model used is the Cat on our GitHub.

https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/WIP/Tridge/TailSitters/Cat

IamPete1, 2020-10-30:
I have just found that we have a pram for this, `Q_TRANS_DECEL` you will have to set it significantly higher than the default.

robustini, 2020-10-31:
@IamPete1 Yes, I tried and setting it to 5 the transition is much closer to the desired point.
However I noticed an unpleasant thing in the SITL, setting this parameter to 0 when I activate RTL the SITL crashes.
Ok that is not an accepted value, but if this happens in the SITL I don't dare to imagine what could happen in reality.
So I ask @tridge to check this thing out.

robustini, 2020-10-31:
Same issue in the Planner SITL and on my Ubuntu.

[image]

IamPete1, 2020-10-31:
Yeah, its a divide by zero, it would not crash the firmware on a real vehicle, but might cause funny stuff.

IamPete1, 2020-11-03:
@robustini can we close this? I think the `Q_TRANS_DECEL` param does what you want?

closed 2020-11-03 by robustini