ArduPilot/ardupilot #15667: [VTOL] Missing RTL_LOIT_TIME in case of quadplane

robustini, 2020-10-28:
I'm finding that with non-vectored tailsitters the critical point during a QRTL is when the drone arrives on the home.
Being pitch inclined, it does not have time to stabilize once it reaches home, and begins to descend losing its attitude.
It would therefore be useful to have the RTL_LOIT_TIME parameter as in the Copter code, so that the tailsitter has time to stabilize the attitude before starting to go down on the home.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2020-10-28:
I'm fairly sure this can be fixed with a larger Q_A_THR_MIX_MAX. I think you have been testing in realflight? Can you share your model and params (I think you already shared with me somewhere, but I cant find them).

robustini, 2020-10-28:
I tried everything @IamPete1 and I never got a decent result.
The model used is the Cat on our GitHub.

https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/WIP/Tridge/TailSitters/Cat

robustini, 2020-10-28:
Q_A_THR_MIX_MAX set to 0.9, nothing has changed (on the sim).
Test with the parameters on our GitHub as a starting point, move the tailsitter away in Cruise, activate Q_RTL and let's change flight mode to RTL, you'll see what happens.

IamPete1, 2020-10-28:
@robustini This is what I get, maybe you have something else? https://youtu.be/evjh0IAwogs

Has to be said the tune is not great.

robustini, 2020-10-29:
@IamPete1 and this result you got it with the parameters attached on GitHub?

robustini, 2020-10-29:
> 
> 
> @robustini This is what I get, maybe you have something else? https://youtu.be/evjh0IAwogs
> 
> Has to be said the tune is not great.

You have activated QRTL from flight modes in the Planner, try setting Q_RTL_MODE with value 1 instead and then selecting RTL in the flight mode and see what happens...

robustini, 2020-10-29:
My result with the default parameters but with Q_RTL enabled: https://youtu.be/uFD9VTtlgUw

robustini, 2020-10-29:
@IamPete1 anyway try as I told you.
If you give QRTLs from flight modes it gets to me well on the home point, try making it transition using RTL and enabling Q_RTL_MODE instead.

IamPete1, 2020-10-29:
I have duplicated that test and do not get the same loss of control.

https://youtu.be/sd-N4HrmnEY

I used these params [CatTailsitter.txt](https://github.com/ArduPilot/ardupilot/files/5460527/CatTailsitter.txt)

robustini, 2020-10-30:
After hours of testing (damn) I found where the problem lies, it's all related to Q_M_THST_HOVER.
In the sim I used hover learn activated because I often fly in Q_Stabilize (even in reality), and this generates the problem in transitions and QRTL.
The automatic learn detects me a value around 0.37, and this raises the problem.
Just manually set it to a value beyond the real hovering point, for example 0.6 and everything works fine.
This means that in my opinion there is still something anomalous, as the high (unreal) hovering point then creates a misalignment of the throttle in QStabilize, which is not good.
The video shows the problem and "the fix": https://youtu.be/7Ya_QdkLD-M

robustini, 2020-10-30:
First transition with Q_M_THST_HOVER at 0.6, the second with 0.4.

[image]

IamPete1, 2020-10-30:
The conclusion is that this seems to be at least in part due to the 'Q_TAILSIT_THSCMX' 'ATT_THR' option. This assumes that a leaned over vehicle muse be going fast and reduces the gain, if not going fast this just results in loss of control.  A possible fix would be to use the target lean angle rather than the actual, but TBH I'm not a huge fan of this method at all. @kd0aij Maybe you have some ideas.

As a result of taking another look at the scaling stuff I have done a small PR that should make it easier to see that is going on in the log, but should not cause any change to flight. https://github.com/ArduPilot/ardupilot/pull/15688

kd0aij, 2020-11-01:
Does the problem go away if Q_TAILSIT_THSCMSK is zero or one?
If ATT_THR speed scaling is causing a problem in auto modes, we should just disable it while in AUTO. Its purpose is solely for manual flight at high speeds in VTOL modes anyway. If you aren't doing that you should definitely turn it off.

kd0aij, 2020-11-02:
Why don't we just force a higher throttle level (than Q_M_THST_HOVER) during the back transition?
Perhaps TK0FF_THR_MAX would be more appropriate.

robustini, 2020-11-03:
> 
> 
> Does the problem go away if Q_TAILSIT_GSCMSK is zero or one?
> BTW, the default is 1 and I don't see it in the parameter list for the Cat, so the ATT_THR speed scaling should not be active at all.
> 
> If ATT_THR speed scaling is causing a problem in auto modes, we should just disable it while in AUTO. Its purpose is solely for manual flight at high speeds in VTOL modes anyway. If you aren't doing that you should definitely turn it off.

Unfortunately it is my case instead, I often fly in Q_Stab, and if a failsafe is activated in QRTL the problem occurs, so disabling it is not the good solution imho.

kd0aij, 2020-11-03:
What values are you using for Q_TAILSIT_GSCMSK and Q_TAILSIT_GSCMIN?

closed 2020-11-30 by robustini