ArduPilot/ardupilot #15672: Equivalent Airspeed and calibrated airspeed

fr-correia, 2020-10-28:
## Bug report

Hi guys, I have been recently looking for the airspeed and how it is usually handled.
So I noticed that we get a **Equivalent Airspeed (EAS)** from the sensors and it is then transformed into a **True Airspeed (TAS)**, using the ratio EAS2TAS, given from our usual barometers.

**So here is my question, and I believe it to be quite relevant.** Our airspeed sensor first provides us an **Indicated Airspeed (IAS)**, and by calibrating the airspeed we find the offset we require to eliminate sensor errors, giving us a **Calibrated Airspeed (CAS)**. 
Well, that is precisely my focus here. For a standard mean seal level we can assume that CAS and EAS are the same, but in the current algorithm we never correct compressibility errors with increasing height! We call it EAS but it really is CAL, I suppose? I would appreciate a feedback on this question! 
Thanks a lot!

**Version**
4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Pixhawk

magicrub, 2020-10-28:
It actually doesn't matter what your altitude is, it's all about the pressure. Airspeed is measured by a differential pressure between a static and dynamic pressure. The difference is your speed. As air gets thinner the measured difference will go down but also your lift. So you need to go faster to maintain the same lift. Those are fairly equal.

Here's a rough example with some made up numbers and assuming zero wind:
At sea level your stall speed is 10 m/s and the measured pressure differential measures 10m/s. GPS groundspeed shows 10 m/s.
At 30k ft altitude your stall speed is 20 m/s and the measured pressure differential measures 10m/s. GPS groundspeed shows 20 m/s.

So, you configure your aircraft once and it works for all altitudes. The effective airspeed auto-magically scales.

@samuelctabor please feel free to sanity check that.

fr-correia, 2020-10-28:
Thanks a lot @magicrub! I think I got it!

closed 2020-10-28 by magicrub