ArduPilot/ardupilot #15679: Fly using barometer

fr-correia, 2020-10-29:
## Feature request

So I usually fly using gps, mainly to the reason of the barometer presenting a drift.
Recently I have been making a focus on this topic, as I would like to use the barometer as primary altitude source to fly. (If taken care of, the baro can present relatively low errors.)

**Problem description**
The problem is, as I've seen in many issues here, the change of the weather conditions. Flying for longer periods, using EK2_ALT_SOURCE as 0, can lead to big offsets. 

**Proposed Solution**
I've also seen a parameter in the baro "GND_ALT_OFFSET", and I suppose that parameter corrects the altitude in longer flights.  For example, if my QFE changes during the day, or if I'm given another QNH, I can calculate the offset using the formula: Offset: 1/LAPSE_RATE * (newQNH - oldQNH), lapse rate being the ISA lapse rate for a standard atmosphere, in this case 1"Hg/1,000 feet. The calculation of this offset is correct, I suppose?
So the main thing is that the external parameter is an altitude, but would be a lot easier to be a pressure, allowing myself and others that desire to fly using barometers to be able to continuously make these corrections.
This is my feature request, an external pressure instead of an altitude correction ;)
Hope I could be of any help, but if I'm talking rubbish, please, feel free to discuss it with me!

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

magicrub, 2020-10-29:
yes, using pressure instead of meters is the obvious better and well adopted method. Meters was chosen a long time ago because it's easier for armatures to understand but pressure is what it should be. Needs a new param.

fr-correia, 2020-10-29:
Thanks Tom @magicrub , I believe I'm on the right track then. Hope I could add something significant to the project!

closed 2020-10-29 by fr-correia

WickedShell, 2020-10-29:
You can adjust the barometer directly, and if you have the data for it it's much better then the `GND_ALT_OFFSET` method, use `GND_ABS_PRESS` and `GND_TEMP` in conjunction to correct the barometer reading. These both must be referenced to your original takeoff location, but if done so and you calculate the correct baro reading for that height you can slew these around correctly.

fr-correia, 2020-10-29:
I see your point @WickedShell, but imagine yourself to be very far away from your original takeoff location. How accurate will our results be if we continue to use ground temperature and pressure at the ground control station? In the case of using the GPS as primary altitude source, the results remain consistent because the altitude is calculated as the altitude provided by the gps subtracting the field elevation, so even if you are very far from home, the vertical position of your reference will remain at a constant elevation, no matter where you are in the planet -> the altitude provided to the EKF also remains consistent. 

We could just update the GND_ABS_PRESS to update our QFE for the pressure for which we have the same elevation from home, but what about temperature? Can we continue to use same temperature from the gcs, even tough we are so far away?

WickedShell, 2020-10-29:
You can apply any correction through the pressure and temperature to correct for the location. Most people don't have access to an easy source of correction data, so making this generally work the same way a Kollsman window hasn't been a priority. You will also end up seeing a number of small errors this way that tend to matter more with flying close to the ground.

With all the above being said though, I only fly `EK2_ALT_SOURCE` 2 and I'm quite happy with that, and that is my general recommendation for how to setup a vehicle.

As a general note with all of these parameters you need to be a bit careful about how fast you tweak them, large step changes can result in the EKF rejecting the baro data.

fr-correia, 2020-10-29:
What about making the altitude from the barometers to be independent from temperature? just like a Kollsman window?
For a given QFE (pressure measured in the airfield) we can easily calculate the QNH if we know the airfield elevation, using ICAO standard lapse rate. Am I missing something here, or do we absolutely require a temperature?