ArduPilot/ardupilot #15683: Copter: Scripting on F4/fmuv3 not reasonably possible?

olliw42, 2020-10-29:
## Bug report

**Issue details**
Trying to use scripting on a pixhawk with 2MB(rev3)/fmuv3. Fullfills the criteria given in the docs of 2 MB’s of flash and 70KB of memory. Loaded with stable ArduCopter 4.0.5, with only most basic setup, so no additional ram snatchers, see parameters:  [ac405-script-issue-params.zip](https://github.com/ArduPilot/ardupilot/files/5460793/ac405-script-issue-params.zip)

Setting scripting to 1 and restarting leads to:
[image]
MAVLink parameters are not completely loaded, one thus has to reset by going through flashing other vehicles.

Going through reduced heap sizes leads to

16348:
[image]

24576:
[image]

32768:
[image]

back to default 43008:
[image]
MAVLink parameters are not completely be loaded.

Here I had no SD card inserted, but the same happens also with SD card inserted.

=> scripting doesn't appear to be possible on F4/fmuv3 boards, even then it is in a most prestine state.

:(

**Version**
Copter 4.05

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
pixhawk, fmuv3

**Logs**
none

WickedShell, 2020-10-29:
It's possible, but it doesn't combine with a number of other memory heavy things. Do you have terrain enabled, or srtl enabled? Both of these eat enough memory that you may immediately be out of memory for scripting on copter.

I will also say that the increased number of bindings does have a negative effect on the amount of memory you need. I know how to fix this, but haven't had the time to do it yet, even though it would be a large boon for memory (and particularly with multiple scripts)

olliw42, 2020-10-29:
freshly loaded stable ArduCopter 4.0.5, and did only most basic setup (frame, acc calib, compas calib, rc calib), no nothing else
so I did not enable anything special myself, the params file you can find in the above

WickedShell, 2020-10-29:
Oh, sorry missed the param files. I don't have a board nearby I can test with, but I suspect if you set the following parameters you will be much better off:
```
SRTL_POINTS 0
TERRAIN_ENABLE 0
```

olliw42, 2020-10-29:
ahhh ... indeed !
this does do the trick, many thx :)

I actually never looked at these params since I never used these things and was assuming that given their cost they wouldn't be enabled per default ... (for them it's also not as obvious as e.g. for batt since they don't hide if not enabled)
maybe it would be appropriate to disable them per default, especially terrain ... or at least a respective comment in the docs might be appropriate

btw, I was a bit puzzled that a heap of 32k wasn't sufficient, given that the param doc indicates 1k as minimum and in the code I think I found something like 8k as minimum.

anyway, I close since it's answered, thx again.

closed 2020-10-29 by olliw42